Ultimaker/Cura #18928: Support structure did not use the layer height of its own base.

richardcdean, 2024-04-22:
### Cura Version

5.7.0

### Operating System

Windows 10

### Printer

Custom

### Reproduction steps

1- Slice
2- zoom into internal structure

### Actual results

Support structure did not use the layer height of its own base.

### Expected results

The support structure must rise from the layer below it.

### Add your .zip and screenshots here ⬇️

[image]
[L-Engraver-Yroller-1.9.zip](https://github.com/Ultimaker/Cura/files/15056053/L-Engraver-Yroller-1.9.zip)

Asterchades, 2024-04-22:
The `Initial Layer Height` setting applies to exactly that - the initial layer, or layer 1 (0 internally). It doesn't apply every time a new feature is started.

The air gap is determined by the `Support Z Distance` setting and its children. You appear to have this set rather high (approximately four layers) which is creating a significant air gap, likely leading to both failed and inadequate supports.

You have also highlighted a `Support Interface` rather than the supports themselves. These have their own settings which can be adjusted independent of the rest of the support structure - though its layer height cannot be changed from the project's global setting.

As you've only provided an STL file rather than a project file there's little further insight I can offer.

GregValiant, 2024-04-23:
Yep.  You need to be in the "Custom" settings to view them.
"Support Z Distance" is the parent and "Support Top Distance" and "Support Bottom Distance" are the kids.  The "bottom" distance applies to any support that builds on the model.

The typical setting would be for "1 layer height" of air-gap between the top and bottom of the support, to the model.  That makes the support easier to remove and with less scarring.
If your Custom printer has dual extruders and you are printing supports with PVA (or some other specialty material that doesn't bond to the main material), then you could set the air-gaps to "0".  That would result in a print that is more dimensionally accurate with a better finish on the first layer above the top of the supports/support interface.
There is a similar setting if you happen to use a raft for a print ("raft air-gap").

I'll go ahead and remove the bug label and close this as it involves a user setting.  Let us know if you feel that is premature.

The Support Top Distance is set to 0.20 (1 layer height).  I've used a Cutting Mesh to change the Support Bottom Distance to 1.0 where the cutting mesh touches the model.
[image]

closed 2024-04-23 by GregValiant