Ultimaker/Cura #18935: Reduced Print Volume In 5.7.0 from 5.6.0 - Ultimaker S3

Kvan-VTC, 2024-04-22:
### Cura Version

5.7.0

### Operating System

Windows 11

### Printer

Ultimaker S3

### Reproduction steps

Printing a model (Ultimaker Project File) close to volume limits works fine in 5.6.0. Same project in 5.7.0 (clearly within limit area) fails notification that it exceeds volume limits.

### Actual results

5.7.0 Failed Volume Limits (Sidenote- increase in print time from 5.6.0 on same file)

### Expected results

Successful print

### Add your .zip and screenshots here ⬇️

[5.6.0 Screenshot 2024-04-22 054641.zip](https://github.com/Ultimaker/Cura/files/15063254/5.6.0.Screenshot.2024-04-22.054641.zip)

GregValiant, 2024-04-22:
Thanks for the report.
I opened the project file in 5.7.0 and didn't have a problem with it.  I'm just looking at the preview though.  I didn't try to send it anywhere.
Someone from the Cura team will take a look.

HellAholic, 2024-04-22:
@Kvan-VTC 
Looking at the screenshots you included, your model is sliced with White tough PLA, but the project file you included has Generic TPU selected so none of the changes you have in the White Tough PLA is included in that project file.
The difference in print time is due to having support enabled/disabled or any changes you might have in your local profile.

[image]

I sliced your model in Cura 5.7.0 with Tough PLA and sent to a UMS3 (printer had black tough PLA loaded and the color doesn't affect the profile). Printer accepted the file and started printing.
[image]
I manually aborted the print after the start sequence finished and the printer was about to print the first layer. And no build volume warning. I do not see an issue relating to Cura here. You can contact the Support on the forums for more help.

Kvan-VTC, 2024-04-22:
Yep, when you send it to print is states it exceeds limits (shown in screen shot). Same project in 5.6.0, prints fine.

 

Keith Van Wemmer

Office: 480.359.7004

Cell: 480.415.3606

 <http://www.vantekconsulting.net/> 

 <https://vantekconsulting.net/> VanTekConsulting.net

 <https://vantekconsulting.net/online_learning/> VanTek Online Learning

Kvan-VTC, 2024-04-22:
You loaded it as a project, I loaded as a model and changed the materials. TPU was 8 hours, the PLA was 4+.

 

If I do as you did, load as a project , then change materials and slice, it look as if it will print (I am printing using the load model using Cura 5.6.0).

 

So why the difference, works on 5.6.0, not on 5.7.0? (rhetorical question).

 

Why, if I load as a project, change materials, then slice, it (appears) to work using 5.7.0? (rhetorical question).

 

Why does changing the material in the project make the print respect the build volume? (rhetorical question).

 

It seems to me these are issues in Cura i.e order of operations so to speak that were not present in 5.6.0.

 

Just observations…

 

Thank you for the response and the workaround.

 

Keith Van Wemmer

Office: 480.359.7004

Cell: 480.415.3606

 <http://www.vantekconsulting.net/> 

 <https://vantekconsulting.net/> VanTekConsulting.net

 <https://vantekconsulting.net/online_learning/> VanTek Online Learning

 

image.png (view on web) <https://github.com/Ultimaker/Cura/assets/28710690/10d00fa4-0feb-4181-9dd6-37be350c909a> 

image.png (view on web) <https://github.com/Ultimaker/Cura/assets/28710690/e8d431e6-0e5c-4bae-a47f-f19334c0e21a>

HellAholic, 2024-04-22:
Cura has 500+ settings that get transferred when you open as project, if you do not want the settings you import as model. So why the difference? In every release more settings are available for users to tweak and fine tune their print, and those settings are not present in previous versions of Cura. So when you use an older version, some of the settings included in the project file are ignored.

Pointing back to the above, a project file has all the settings included for that model + nozzle + material + ... when you switch material, the changes are not transferred. 
Easy example, try using Generic Tough PLA to slice a model at a certain layer height, then switch to Ultimaker Tough PLA and slice using the same layer height, why is there a difference of 3 hours in the print time? Printer profiles, material settings,... all contribute to how it is being handled.

If you still believe the issue is caused by Cura, please include the project with configuration that you sliced and sent to printer in 5.7.0; in addition try saving the slice file to disk and include that one as well.