Ultimaker/Cura #18939: FLSUN Super Racer (gcode flavor please add klipper as I now have the speeder pad and it controls the printer running klipper)

techietrucker, 2024-04-23:
### Is your feature request related to a problem?

not printing right becuase I can't slect klipper as gcode flavor, just marlin which the printer is no longer running on.

### Describe the solution you'd like

it's all in the title:)

### Describe alternatives you've considered

using a different slicer which supports klipper on the super racer

### Affected users and/or printers

FLsun SR is affected

GregValiant, 2024-04-23:
I think there is some work being done on adding Klipper.  It's probably at least a year out though.
There is a Klipper plugin in the Marketplace that you could try.

techietrucker, 2024-04-23:
Once I installed the plug-in I still did not have that option
unfortunately, unless I'm doing something wrong

GregValiant, 2024-04-23:
It's all there is.
None of the Ultimaker printers use Klipper.  They are not likely to pay their people to write the code to incorporate something that their own customers can't use.
Check around the internet.  I don't know about your specific printer but others get Cura to work with Klipper.

Asterchades, 2024-04-24:
Klipper printers function near-enough perfectly when configured as Marlin under Cura, in spite of the author's efforts to needlessly break global compatibility. I've been using it exactly as that since 4.6 (experimenting as early as 4.2), with the only change I've needed to make being to create an M900 macro for Pressure Advance - but this is not default functionality as provided by Cura, rather something I've added via a plugin.

The only built-in functionality that would not be appropriately transferred is going to be pausing (such as the `Pause at height` script), jerk configuration (Klipper uses `Square Corner Velocity`, which is not recommended to be changed from the default 5 anyway), and filament changes (the `Filament Change` script). These would require macros to alias their Marlin command equivalents (though an M25 pause is already usable if using the Virtual SD Card feature).

Features such as bed levelling or configuration profiles are also not natively supported, however these are also not native Cura features. These are all hard-coded into the various 3rd party definitions which include them, and would require either being manually rewritten or a Klipper macro rather than simply changing the GCode flavour.

Naturally you would also want to add either the Moonraker or OctoPrint plugin (depending on your host environment) from the marketplace to allow network connection to your printer, but even without that you can still manually upload the created GCode files to the host.

_The Klipper Settings plugin doesn't add Klipper as a GCode Flavour as it's largely unnecessary. It instead adds an extra printer setting section (along with Walls, Infill, Supports, etc) which adds additional, Klipper-specific settings - such as pressure advance, tuning towers, velocity limits, input shaping, etc._