Ultimaker/Cura #18942: arachne engine fails horribly to fill zones

petaflot, 2024-04-23:
### Cura Version

Prusa Slicer Version 2.7.1

### Operating System

Gentoo GNU/Linux

### Printer

custom build

### Name abnormal settings

no crash ; jut bad result

### Describe model location

dead center

### Describe your model

it's a box to hold marking tools (to hammer letters in metal).

### Add your .zip here ⬇️

yes, I know it's not actually cura but the engine shouldn't matter depending on the GUI used.

I wouldn't bet it's a but, but I really couldn't figure a way to set the parameters right. it may not be obvious on the photos, but some lines start and then simply vanish into thin air. 

[arachne_engine_fill_issue.zip](https://github.com/Ultimaker/Cura/files/15082437/arachne_engine_fill_issue.zip)

### First try

* two shells

### Second try

* 4 shells (on the black only, I merged the gcode manually, see arachne_on_logo_fail_theoretical2)
* max perimeter width: 65%
* min feature size: 15%

petaflot, 2024-04-23:
oops sorry not a slicing error, but a bug report

GregValiant, 2024-04-23:
Thanks for the report.
It looks to me that your "Initial Layer Height" is interfering because of the way the model is designed.  With the Initial Layer Height at 0.15 (same as the layer height) the problem goes away.
Here is layer 2 of that project.
[image]

So this appears to be a setting issue rather than a bug.  I'll remove the bug label but leave this open for a while so you can respond here.

petaflot, 2024-04-23:
@GregValiant your reply is not making sense, it seems you have not understood the issue : if you take slice of my model between 0.3 and 0.6mm, you will see it is solid. 

also, regarding this "second-layer" issue you should have answered not here but on #18943 (they are quite unrelated issues). and it just happens to be the second layer precisely because of the way the model is designed : if I wish, I can make this bug (rather, a limitation that shows a problem in the slicing algorithm and impacting probably both the slicing performance and the print times) appear on any layer, any number of times.

petaflot, 2024-04-23:
with 0.15mm slice thickness, check layer number 4

petaflot, 2024-04-23:
> Thanks for the report. It looks to me that your "Initial Layer Height" is interfering because of the way the model is designed. With the Initial Layer Height at 0.15 (same as the layer height) the [...]

replying here: #18943

GregValiant, 2024-04-24:
This is another report that should go to Prusa.  If you are having a problem with Cura then report that here.  

I'm going to close this one.  All of this cross-referencing (with no project file) is confusing.

The Cura Engine back end is a separate Git [HERE ](https://github.com/Ultimaker/CuraEngine/issues) and a bug report,  complaint, or discussion regarding the Cura Arachne Engine (the Back End) should be made there.  If you want to report a bug in regards to Cura settings (not PrusaSlcer) or the GUI "front end" then this is the place.  Try again and please post a Cura project file.  Without knowing how you've set things up in Cura there can't be an investigation.

Understand that I'm not a member of the Cura team and I don't work for UltiMaker (or Prusa).  I am an investigator and a contributor so I have some knowledge on how things work.  You need to be concise and provide all relevant information that may assist in an investigation.

closed 2024-04-24 by GregValiant

petaflot, 2024-04-24:
@GregValiant thank you