Ultimaker/Cura #18951: Running Cura 5.7 slicing failed

tomlogan1, 2024-04-24:
### Cura Version

5.7

### Operating System

Windows 11

### Printer

Ender 3 Pro 4.2.7 board 

### Name abnormal settings

no, just moved from 4.13.1 to 5.7

### Describe model location

no change

### Describe your model

Filament rack 

### Add your .zip here ⬇️

[CE3PRO_Filament_storrage_stl.zip](https://github.com/Ultimaker/Cura/files/15101945/CE3PRO_Filament_storrage_stl.zip)

GregValiant, 2024-04-24:
Thanks for the report.
I didn't have any trouble slicing it in Win10 and 5.7.1.
Try right clicking on the Cura desktop icon and select "Run as Administrator" and see if it makes a difference.  Let us know.

I'll mark this as a duplicate.

tomlogan1, 2024-04-25:
> From "GregValiant" ***@***.***>
> To "Ultimaker/Cura" ***@***.***>
> Cc "tomlogan1" ***@***.***>; "Author"
> Date 4/24/2024 6:24:44 PM
> Subject Re: [Ultimaker/Cura] Running Cura 5.7 slicing failed (Issue
> #18951)

I rotated the object and was able to slice it successfully.  Saved the 
gcode. Reverted the object to its original position and had the same 
error.

This email has been checked for viruses by Avast antivirus software.
www.avast.com

HellAholic, 2024-04-25:
Just a passing by comment:
Checked the model, it looks like an issue we're investigating.
Could slice with 0.2 layer height or running as admin in 0.48 layer height (selected originally in the project).
The error shown in the logs for the crash is:
`[error] The file %s could not be opened for writing.`

tomlogan1, 2024-04-25:
Just setting up for 0.6mm nozzle and got the settings from another user.
Strange, though, it would post the error if the piece pointed left to right
but not when pointed top to bottom.

Thanks for your insight.

HellAholic, 2024-04-26:
Quick update,
Found the culprit, should be resolved with CURA-11821