Ultimaker/Cura #18953: It is impossible to slice...

awiscan, 2024-04-25:
### Cura Version

5.5.0 — 5.7.1

### Operating System

Windows 11

### Printer

Anet A8, Creality Ender 3 v3 KE

### Describe your model

Groot

### Add your .zip here ⬇️

1. The error "Cannot be sliced" appears during the initial loading of any model (stl).
2. When changing the size of the walls or the thickness of the layer, the slicer automatically starts again and cuts the model normally.
3. It is observed in versions from 5.5.0 to 5.7.1.
[UCP_Groot hello sitting.zip](https://github.com/Ultimaker/Cura/files/15107040/UCP_Groot.hello.sitting.zip)

awiscan, 2024-04-25:
P.S. 
[image]

HellAholic, 2024-04-25:
Hey @awiscan 
Could you please save the file as a project instead of universal cura project (UCP)?
[image]

The issue is probably somewhere in your settings if you cannot slice any model and a project file allows us to find the faulty setting quickly. Universal project is printer agnostic so it does not save all your settings.

awiscan, 2024-04-26:
Hello. Thanks for the reply. The project file is attached.

HellAholic, 2024-04-29:
Hey @awiscan it is most likely a setting was overwritten from a project file you've opened that caused the issue. So best recommendation would be going to %appdata%/roaming/cura and removing the configuration folders in there.
These will remove the potential corrupted settings/definitions from your device.

What you can do:
1. Go to the %appdata%/roaming/cura
2. Back up the contents (any version you see a folder in there 4.x or 5.x) -> make a zip and store somewhere as backup
3. remove all the contents of the %appdata%/roaming/cura (any version 4.x or 5.x)
4. Start Cura again
5. Add your printer
6. Attempt to slice

If the process didn't succeed, you can restore from the backup.

Also note this process will remove all user profiles/custom settings/installed plugins/etc.