Ultimaker/Cura #18954: Adaptive Layers causes Pillowing

LilBub, 2024-04-25:
### Cura Version

5.7.0

### Operating System

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

I've encountered this for a few prints but I haven't been able to remedy it on my own.  This bear trophy has a flat surface where a placard would go.  You can see the pillowing on that surface and the top of the bear. I'm attaching a photo and Project file.
[UMS5_Black Bear Trophy (half scale) ufp.zip](https://github.com/Ultimaker/Cura/files/15112632/UMS5_Black.Bear.Trophy.half.scale.ufp.zip)

### Actual results

As stated.  (Pillowing)

### Expected results

Smooth surfaces

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-04-25:
Thanks for the report.
That looks like the result of seriously wet filament.  Little steam pockets form and blow the skins out.
It also looks like you are under-extruding by a lot...maybe 30%.  There should not be any gaps between the extrusions of a skin.
Try a different roll of filament and you need to make sure that Cura has the correct filament size set and that it is not set to "Volumetric" unless the printer is also set to "Volumetric".
I'm confident this isn't a bug but rather a problem with the filament and/or printer.  I'll remove the bug label for now (it can always be put back on).

LilBub, 2024-04-25:
Interesting since it's a new spool of material and we're strictly using Ultimaker S5 default settings, asides some tweaks to the supports.  We can certainly throw it in the filament dryer and try again.

GregValiant, 2024-04-25:
I had the problem with a new spool of Sunlu filament.  I don't know how it got that way, but it's the only filament I ever suffered the blowouts with.
If it was any other printer I'd recommend calibrating the Extruder.  UM's don't have the option, but something is going on to cause the under-extrusion.
You can see here that the extrusions certainly aren't welding together.  You shouldn't be able to see a previous layer skin through the current layer skin.
[image]

LilBub, 2024-04-25:
Well they do use water in the cooling lines when they make filament.  Ideally, we should be drying before using any materials.

For comparison, I have another one printing without Adaptive Layers checked on

LilBub, 2024-05-02:
Checking back in - we put the materials in the PrintDry Pro for a bit and tried again.  Much much better results.
It's the filament, not Cura!
We usually don't see damp filament but in talking to the student workers, it appears they have skipped the 'dry before use' and figured it didn't need it with new spools.

GregValiant, 2024-05-02:
It's good that you got it sorted out.  I've put around 75 rolls of filament through my Ender and out of all of them there was only the one "wet" roll.  The failure mode is highly distinctive.

Steam explosions...just another bizarre aspect to 3D printing.

I'll go ahead and close this then.

closed 2024-05-02 by GregValiant