Ultimaker/Cura #18955: Add option for wall ordering "inside to outside to inside", similar to inner/outer/inner wall ordering in Orcaslicer

persano, 2024-04-25:
### Is your feature request related to a problem?

To improve outside quality and also accuracy but without reliying solely on outside to inside, I propose the incorporation of a third option, which already exist in another slicer.

### Describe the solution you'd like

This will give better outside walls and gives better results outer/inner only from orca slicer. You need 3 >= walls for it to function, if less walls, it will work as the outer/inner option we already have.

### Describe alternatives you've considered

None. 

### Affected users and/or printers

It will affect all printers. I am not saying this should be the default option, but we should have the possibility to choose it.

### Additional information & file uploads

[image]
[image]

KimmoHop, 2024-07-21:
I wonder if new plugin system could override wall order setting, or if complete overhaul of current implementation is needed...

persano, 2024-07-21:
> I wonder if new plugin system could override wall order setting, or if complete overhaul of current implementation is needed...

I do not know if that is possible. I work as a dev and would like to dig down on the cura engine, but I have never tested it. If I have some free time, I can test this by myself. But for now, I do not see me making a plugin or any merge request for adding this into the slicer (would love to, but I really do not have any free time to make it a reality)

Recticle, 2024-09-26:
I really want this too, I hope they implement it soon.