Ultimaker/Cura #18956: Slicing Failed

ehei, 2024-04-25:
### Cura Version

5.7.0

### Operating System

Windows 11

### Printer

Creality CR-10 Max

### Name abnormal settings

Using Dynamic Quality setting

### Describe model location

Viewing from straight above

### Describe your model

Several individual Gridfinity models added to one file

### Add your .zip here ⬇️

[CCR10MAX_BETA_2x4_Catch_All.zip](https://github.com/Ultimaker/Cura/files/15119295/CCR10MAX_BETA_2x4_Catch_All.zip)

GregValiant, 2024-04-25:
Thanks for the report.
I was able to slice this in 5.7.0 and 5.7.1 on a Win10 system.
Of the 4 models on the build plate, 3 of them aren't watertight.  At least one of the others has errors as well.  That needs to be addressed because models that need repair can lead to untrustworthy gcode.

Someone from the Cura team will take a look.  I'll mark this as a duplicate report.

ehei, 2024-04-25:
Great,  thanks.  I was able to remove several models and get it to slice, so I'll have to look at the ones I downloaded and give the creators some feedback ;)