Ultimaker/Cura #18959: Cooling fan not starting

Jeffrey65511, 2024-04-26:
### Cura Version

5.7.0

### Operating System

windows 11

### Printer

Creality Ender 3 V2

### Reproduction steps

2 weeks ago I installed a Creality Spider Extruder Pro Kit and at the same time, I updated Cura to version 5.7.0.
Immediately anything I tried to print would display signs of over-extrusion it has taken me until now to resolve the several issues that occurred.

1. The spider book states the steps/mm for the e-motor incorrectly at 424.9. Through trial and error, I have found 375.0 to work properly.
2.  Cura 5.7.0 does not control the cooling fan, giving the impression of over-extrusion, which I did not notice for several days before having reverted to  Arachne Beta Engine 2. Since then the  printer has been working properly

### Actual results

M420 S1 : Auto Level

### Expected results

The printer is now behaving as it should and I am getting lovely results.

### Add your .zip and screenshots here ⬇️

none

Jeffrey65511, 2024-04-26:
I have just tried out Cura 5.7.1 and it still does not control the cooling fan.

GregValiant, 2024-04-26:
Thanks for the report.
Cura uses "M106 S" as the command for the layer cooling fans.  If there is more than one extruder then the first extruder should be Fan 0 and the second extruder should be Fan 1.  On a multi-extruder printer, Cura will add M106 S255 P0 to indicate the fan for extruder 1 should go to 100% duty cycle.  If you did not make any firmware changes then that should work.  You can also open the gcode in a text editor and search for M106.
On a single extruder printer the command would be "M106 S255" without the P parameter.  The printer will assume that without the P that Fan 0 should be turned on.

Go into Manage Printers and then Machine Settings.  On the Extruder Tab there will be a box for the "Cooling Fan Number" that relates to that extruder.  On a single extruder Ender 3 it should be Fan 0.
If you have Pronterface installed you can try sending "M106 S255 P0" and then "M106 S255 P1" and see if one of them turns the fan on.  Both M106 S0 and M107 turn the fan off.

I'll leave the bug label on this for now but it is likely a setting problem.

GregValiant, 2024-04-29:
Did you get a chance to take a look at the fan settings in Cura?

closed 2024-05-14 by github-actions[bot]