Ultimaker/Cura #18964: Slicing error.  Unable to slice

salocius, 2024-04-26:
### Cura Version

5.7.0

### Operating System

Windows 10

### Printer

Ender 3

### Name abnormal settings

Using normal profile.  Slices OK with other models.  Just crashes on this particular model

### Describe model location

Yes just moved to the left from centre and it now slices OK

### Describe your model

This is modified from a previous version.

### Add your .zip here ⬇️

[thin ring 18mm (1).zip](https://github.com/Ultimaker/Cura/files/15134172/thin.ring.18mm.1.zip)

GregValiant, 2024-04-26:
Thanks for the report.
I didn't have any trouble slicing it and I'm also on Windows 10 with an Ender 3.

Someone from the Cura team will take a look.

HellAholic, 2024-04-28:
This is another instance of the admin bug, so the current workaround is Running Cura as Administrator.
Duplicate of:
https://github.com/Ultimaker/Cura/issues/18835