Ultimaker/Cura #18966: Imported objects in 3MF files' positions are changed

ansonl, 2024-04-26:
### Cura Version

5.7.1

### Operating System

Windows 10

### Printer

Any printer

### Reproduction steps

Import a 3MF file with 2 objects with original positions that interlock.

The original object positions should be retained so that the intended design positions are available to the user if the print requires exact positioning of 2 or more parts. 

In Cura 5.7.1, the second object seems to be pushed over to the side since the software seems to automatically reposition and they overlap in extent. 
In Cura 5.4.0, the import worked fine and original positions were kept. 

### Actual results

The imported objects' original positions are changed during the object arrangement.

Below is the screenshot after importing a file into Cura 5.7.1
[image]

### Expected results

The import object's original positions are kept until explicitly moved by the user. 

Below is a screenshot of the same file imported into Cura 5.4.0
[image]

### Add your .zip and screenshots here ⬇️

An example 3MF file with 2 objects. 
[MD-sqrt-dual-v1.zip](https://github.com/Ultimaker/Cura/files/15135305/MD-sqrt-dual-v1.zip)

HellAholic, 2024-04-28:
Hey @ansonl ,
There was an issue where the import models only would still transfer the positional data and additional settings through which was fixed in 5.7.0;
if you open a 3mf file as project, it should retain all the information related to the position, material, settings, while importing as models should only import the model data itself and no additional settings.

There is now an in between option which is called Universal Cura project which holds the model positional data but you can skip the profiles/settings/etc.

If your report is regarding the import models only function, then the prior behavior you've described was a bug (dressed as a feature). Still you can select the parts and choose Merge models and they should fit back in. But remember to assign the desired extruder to each model before the merge, otherwise you would have to unmerge, and then merge again.

ansonl, 2024-04-28:
@HellAholic, I was able to restore the positions with merge models. The 3MF file that I dragged into Cura was a bare 3MF file exported from Blender with the 3MF plugin. This file did not have any print settings data and Cura did not ask prompt me to import either the "project" or "models only". 

There was no indication that the model had positional data that repositioned upon import and I only knew that the positional data was there because I knew from before. 

Does this seem like a bug like described because the model is a bare 3MF with only positional data and there was no option to import it as a project or model only? There's no indication in 5.7.1 that the positional data existed so most users wouldn't know to select all and merge models unlike keeping the positions intact like previous Cura versions.

HellAholic, 2024-04-29:
Scenario you're describing feels like a feature gap to me personally. I'll add a task to discuss this with the team. It can be seen as either a bug for missing a case when opening bare 3mf or as a feature to allow the models to retain their positional data if there is no printer data within the 3mf.

Internal reference: CURA-11874

ansonl, 2024-04-30:
@HellAholic Understood that makes sense why the default behavior was changed. 

I would like to suggest that an alert/popup could be shown on import of a 3MF with existing position data not changed. This way the default repositioning would keep the objects close to the buildplate but also alert the user that original positions exist and Merge can be used to restore the original positions. 

Is there a difference between Group and Merge objects function besides restoring original position data (for merge)?

HellAholic, 2024-05-01:
I proposed a solution similar to your suggestion but it will add a step for all users importing a 3mf and therefore not ideal, so a final solution could be in the shape of a plugin in the marketplace. This is mainly because different 3D model applications have different behavior when it comes to positional data and size. 

For the Group vs Merge -> Group uses the current position of the items inside Cura and Merge tries to use the positional data from the files. Outside of that, I'm not sure if they have any other differences :P

ansonl, 2024-05-02:
@HellAholic I tried merging a 3MF that had position data today and the position of the objects after Merging were off. This problem only happened on some files and not others. The files were all created the same way and I can't tell why the merged position is off on some of them. Merge also rotates all files by 90 degrees on the X axis for some reason but that's not the position issue.

[image]

[image]

I confirmed that the 3MF file has the correct position data by importing it into Blender as seen in the screenshot below:
[image]

The 3MF file is 26MB which is over the github limit of 25MB so I uploaded it for testing on google drive: https://drive.google.com/file/d/10E2R-l785HEOxNsv98ZotDAP9GXQS6Ha/view?usp=sharing

Do you know what might be causing this and how I can keep the original positions?

HellAholic, 2024-05-02:
Hey ansonl,
I'm not really a blender expert :P but I've seen this type of issues for exporting models from blender since you can define the z axis as being forward or up, but the z they use is not necessarily the same z in other application. Not 100% sure, so recommend playing around with the export settings and doing some search in similar terms for blender export.
If it turns out to be related on something in Cura, let me know then I can pick things up from there.

ansonl, 2024-05-02:
@HellAholic The object normals seem exported correctly and it is manifold. I tried opening it in Cura 5.4.0 to see if there was a difference and there's some strange behavior.

When I import the file to Cura 5.4.0, the objects' positions are kept like we expect below:
[image]
When I select both objects and right click -> Merge objects, the second object is consistently offset in the X direction:
[image]

So it seems like the import (by drag) file and keep original object positions behavior is different from the restore positions effect of Merge objects. Since the import and keep positions feature is removed in 5.7.0, I can't test getting the correct position back the original import way in 5.7.

HellAholic, 2024-05-03:
@ansonl I see the same offset issue in 5.4 if I select all models and center them to the build plate, since the models initially load outside of the build plate. So I wouldn't say the offset is a regression from previous behavior (comparing to 5.4)
It also does the flip when you do the center. So I believe the center selected doesn't play nicely with the model you've exported in this case.

[image]

Regardless, thanks for the deep dive :) there is probably a feature gap somewhere in there but it's hard to define at the moment. So if you're willing to investigate a bit more, I recommend trying different combinations with simpler models (start with cubes or disks and gradually make it more complex). That way it's easier to identify each component and make a clear case.

I do have a workaround for you in this case (or any other model you export that has the same behavior).
1. Add the model (drag & drop or open via file menu)
[image]

2. When you see the model parts on the build plate press "Ctrl + Z" which shifts the model to the original state outside of build plate.
[image]

3. Assign the extruders to the model parts (optional)
[image]

4. Select all models > Merge selected
[image]

5. Center the merged on the build plate
[image]

You can either assign the extruders based on the object list or unmerge after center and assign it then.

[image]

Results in the following after slice:
[image]

There are a couple of errors/warnings when trying to slice regarding the model (overlapping faces)

ansonl, 2024-05-04:
@HellAholic Wow, the workaround you mentioned works perfectly! So I guess Cura 5.7 is auto adding a reposition "action" right after importing but luckily it can be undone if we know that it is seen as a normal action. 

The new gradual flow setting has improved the print quality on my dual extruder UM2+ so I'm happy to be able to use 5.7. 

In the past I used Group after selecting all objects and before moving the group to the center as the objects where already at the correct positions. Group seems to work as before after the Ctrl+Z and Ctrl+A you mentioned without the need to rerotate the object so I'll have to experiment and try Merge on some other models to see if I can find out what causes the offset.

ansonl, 2024-05-06:
@HellAholic I was able to reproduce Merge objects issue with a simple test 3MF with 2 cubes (attached) positioned at an angle next to each other with 3 different setups:

1. The bare file itself
2. File imported into Bambu Studio > grouped together > exported
3. File imported into Bambu Studio > not grouped together > exported

All 3 files have the original position data. However when importing the files into Cura and merging objects, files 1 and 3 end up with the cubes blown up in size with original position lost. File 2 has the objects put into a 3MF group that is recognized by Cura and the object positions are kept until ungrouped and merged again. The end result with the cube blown up after merging occurs with all 3 files.

The units should be set correctly to mm in the 3MF files and the cubes are initially imported at the correct size until merged. 

I have the scale large/small models option unchecked in my Cura settings
[image]

correct position
[image]

after merge
[image]

2 cubes with 8 vertices each in blender with unit scale set for export
[image]

[cubetest.zip](https://github.com/Ultimaker/Cura/files/15215938/cubetest.zip)

HellAholic, 2024-05-08:
Quick update,
Added a ticket to the backlog for investigating the merge behavior and hopefully improving it.
Internal reference II ( the return of the Cura ): CURA-11896

gh13, 2024-05-08:
I don't know if it's relevant here but I've had similar issues when opening 3mf files that use the SpoonTabs plugin which was raised with the dev & discussed here [https://github.com/5axes/SpoonAntiWarping/issues/8#issuecomment-2097473923](url)
The dev has come up with a a workaround until the issue is resolved which is once you have created your layout, select all the models & use the Mesh tool 'Reset origin to centre of mesh' before saving

fleutot, 2025-07-29:
I currently don't know how to print with multi-material, because of this issue. This is my model:

[image]

This is what importing with 3mf looks like in Cura 5.6:

[image]

And in 5.10.1:

[image]

All parts drop their Z position.

I understand that this is in your backlog, but surely there must be some work-around? I've tried STL (cannot split it back in Cura), STEP (not supported), export as individual parts, ... Surely, this kind of prints cannot be broken for so long, there must be a way?

HellAholic, 2025-07-29:
If you can group the models before exporting the 3mf, you can then load it in Cura, make sure Drop to build plate is disabled (in preferences and on the printer when selecting the model), then ungroup them. This should keep them in place.

I don't have your specific model so I'm suggesting a generic workflow:
- Select all models
- Uncheck the drop to build plate
- Press Ctrl+Z
- Should place them in their original position from the 3mf