Ultimaker/Cura #18967: Failed to slice an .stl that seems to be tested

Dpariet, 2024-04-26:
### Cura Version

5.6.0

### Operating System

windows 11

### Printer

ender 5plus

### Name abnormal settings

no

### Describe model location

no it still doesnt slice after rotating and changing location on build plate

### Describe your model

It is water tight and I tried using mesh tools (not sure exactly how) but it didnt fix the issue

### Add your .zip here ⬇️

[bltouchmount_4040_fan.zip.zip](https://github.com/Ultimaker/Cura/files/15135648/bltouchmount_4040_fan.zip.zip)

GregValiant, 2024-04-27:
Thanks for the report.
A lot of work was done on the slicing errors that occurred in 5.4-5.6.  I suggest you install 5.7.1 and see if the problem has been fixed.

I'll mark this as a duplicate and go ahead and close it.  If the problem persists in 5.7.1 then please make out a new bug report.

This is your model sliced with 5.7.1.
[image]

closed 2024-04-27 by GregValiant

Dpariet, 2024-04-28:
5.7.1 seemed to have errors with some of the things i was using, ie,
uploading to my sonicpad directly from cura so

GregValiant, 2024-04-28:
But 5.6.0 has the slicing errors.  They were well known and addressed in 5.7.0 and now 5.7.1.
You can go back to 5.2.2 which was pretty error free.  I think the biggest difference moving on from 5.2.2 was advances in Tree Supports.
I have no idea if it will work with your other hardware because I'm unfamiliar with what a "Sonic Pad" is or does.