Ultimaker/Cura #18968: Very slow to load initial STL file in Ubuntu 24.04

StormTrooper, 2024-04-26:
### Cura Version

5.7.1

### Operating System

Ubuntu 24.04

### Printer

Ender 3 Pro

### Reproduction steps

After running Cura if you open any STL file it takes around 80 seconds for a simple file to load. Opening any other file after that (even the same file) and it's instant. This only occurs on the first load of a file. 

I've also only noticed this issue after upgrading from Ubuntu 23.04 to 24.04

### Actual results

STL file take 80 seconds to load. 

### Expected results

Should load immediately. 

### Add your .zip and screenshots here ⬇️

Not sure if the below errors are related as they occur before I load any STL file:

Gtk-Message: 23:33:29.870: Failed to load module "canberra-gtk-module"
Gtk-Message: 23:33:29.874: Failed to load module "canberra-gtk-module"
qt.qml.typeresolution.cycle: Cyclic dependency detected between "file:///tmp/.mount_UltiMaug3H2h/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_UltiMaug3H2h/share/cura/resources/qml/Actions.qml"

** (UltiMaker-Cura:68839): WARNING **: 23:33:56.193: atk-bridge: get_device_events_reply: unknown signature
Opening in existing browser session.

[insert.zip](https://github.com/Ultimaker/Cura/files/15135753/insert.zip)

didli, 2024-04-30:
Could you try to rename the 'plugins' folder in your /home/$USER/.local/share/cura/5.7/ and restart Cura to try again ?
I had the same problem (loading a .stl took around 93 sec) with Fedora 39 and since Cura 5.7.

StormTrooper, 2024-04-30:
Tried renaming the plugins folder, and also tried renaming 5.7 folder. No effect from either. 

Had a look at the cura.log and seeing the same thing as you. Tries for a number of seconds to connect to the printer via serial port. 

[cura.log](https://github.com/Ultimaker/Cura/files/15169078/cura.log)

StormTrooper, 2024-04-30:
Duplicate of https://github.com/Ultimaker/Cura/issues/18882

Disabling USB Printing fixes the issue.

didli, 2024-04-30:
Thank you StormTrooper for the duplicate reference. Strange that it only occured for me beginning with 5.7 version.  
Anyway, for the people with the same problem, click on the marketplace button and see below : 
- - - 
[image]
- - - 
[image]
- - -

frostworx, 2024-05-09:
Thanks @StormTrooper for the hint!

and thank you so much for the detailed instructions @didli 
I barely use cura and would not have found the settings otherwise

I can confirm this fixes the issue with loading stls for me as well.

bmv234, 2024-05-27:
After upgrading to Ubuntu 24.04, I am having the same issue. Disabling USB Printing fixed it for me as well.

CrashTD, 2024-05-28:
Can confirm the same on Arch Linux. Disabling USB Printing helps, but this is of course no permantent solution. Can't print if Cura can't talk to the printer..

So i tried the "USB Printing NG" plugin from the marketplace and that seems to be indeed fixing my problem.
You however need to set `CURA_BAUDRATE` and maybe `CURA_COMPORTS` environment variable.

In my case it was enough to start cura like this:
`CURA_BAUDRATE=250000 cura`

fabiovix, 2024-10-22:
I can confirm the same issue here... and disabling the USB Printing worked for me on Cura 5.8.1 + Ubuntu 24.04.1.

markjuggles, 2025-01-19:
Same with Cura 5.9.0 on Zorin 17.2.  Thank you Didli.