Ultimaker/Cura #18970: Tree support "Infill" going through model

jumper4412, 2024-04-27:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Creality Ender 3 V3 SE

### Reproduction steps

Added a sphere in mid air to test supports
enabled tree supports with 10% density and lines pattern

### Actual results

Two support pattern lines are going straight through the model

### Expected results

The support pattern lines ended going through when the interface started, should not start again after the interface ended

### Add your .zip and screenshots here ⬇️

[Curasupport.zip](https://github.com/Ultimaker/Cura/files/15139060/Curasupport.zip)

jumper4412, 2024-04-27:
[image]

HellAholic, 2024-04-28:
Hey @jumper4412 ,
Please attach a project file so we can investigate the issue. A Gcode just shows the same information as your screenshot in this case and cannot help with identifying settings/parameters that cause the issue.

jumper4412, 2024-04-28:
Sorry, forgot about the project file! I dont have the file from the original request, but i was able to reproduce it, screenshot and files attached
[image]
[Curasupport2.zip](https://github.com/Ultimaker/Cura/files/15142639/Curasupport2.zip)

HellAholic, 2024-04-28:
Thanks for the file and report, I can reproduce the issue.
If I reset to default settings for the layer height, then the support lines don't go through. So without an investigation, the suggestion would be to take a look at the print settings (combination of tree support and the settings) to see if anything pops out. I'll add a ticket for investigation. 

[image]

If I drop the model to build plate, same settings don't result in the issue.
Or if I select normal support.

jumper4412, 2024-04-28:
Many thanks!

HellAholic, 2024-04-29:
Internal reference: CURA-11875

ThomasRahm, 2024-06-03:
That is a interesting bug. 

Issue lies in the function `filterFloatingLines`. The support lines around the object are not be supported, causing the hole in the support area to be removed. Seems there is a check missing to ensure that there is no model inside the hole that is removed. Basically there are two possible solutions.
1. Keep the holes always if there is a model area inside of them. Solves this, but can cause floating lines if infill density is 0%. Causes support to be kind of useless here in this case with 0% support density. I don't like this.
2. Only keep holes with model areas inside it if the support infill rate is larger than 0. Avoids above issues, but I have no idea if the support areas are used for travel/retractions down the road (potentially causing other issues as a model part may be inside the support area). I like that even less.

I think a proper solution would be to generate a different density support area below the hole. I do this now in my cradle branch, but I don't think that this branch will be good to merge before the next version.
<details>
<summary>Example of a different density support area below the hole</summary>

 [image]

</details>

@HellAholic 
Here is the commit to my cradle branch that fixes this (approach 1): https://github.com/ThomasRahm/CuraEngine/commit/ae2553271a89584285a616413f32092770ef07b8
I have seen that a 5.7.2 is planned, so I am not sure if I should be doing a pull request to fix this in the 5.7 branch or main.
If you want this fix liked above as a pull request please let me know if it should target main or 5.7.

@jumper4412
Now that I know what caused this I can describe a workaround for this issue: One can ensure the hole in the support structure gets connected to the outside of the support structure. E.g. by placing a support blocker.

[image]

nallath, 2024-08-02:
Fixed for 5.8

closed 2024-08-02 by nallath