Ultimaker/Cura #18972: Sorry, you have been blocked You are unable to access ultimaker.com

kos19191, 2024-04-27:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

kingroon kp3s

### Reproduction steps

registration in the application
Immediately after installation I cannot register in the application

### Actual results

Sorry, you have been blocked You are unable to access ultimaker.com

### Expected results

registration in the application

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]

fieldOfView, 2024-04-28:
This will not fix your access to ultimaker.com, but you do not need to be logged in in Cura to be able to use it.

HellAholic, 2024-04-28:
Sorry but we cannot help you here from Cura repository. Whether it was due to your activity, regulation changes, or an issue of the website, it is outside of Cura.

closed 2024-04-28 by HellAholic

kos19191, 2024-05-23:
Can you tell me where I should go with this question?

fieldOfView, 2024-05-23:
Again, you do not have to create an account or log in to use the application.

ZEX-SPb, 2024-05-24:
Что делать, если при входе в аккаунт Cura написано: "Sorry, you have been blocked
You are unable to access ultimaker.com"?
[image]

HellAholic, 2024-05-24:
@ZEX-SPb I've removed your IP address due to security reasons for your own benefit. Please do not share your IP on public places.

Regarding your lack of access, you're attempting to access the Ultimaker.com from a country that has been banned from accessing the Ultimaker website. As such, we cannot help you with this issue. It is out of our hands.

nick140891, 2024-11-04:
Подтверждаю. Данная проблема существует и в настоящее время. Дискриминация по признаку местонахождения человека - это, мягко-говоря, крайне не дружественный жест. Прошу исправить эту несправедливость.