Ultimaker/Cura #18975: 3DBenchy.stl スライス結果が？

KouMRT, 2024-04-28:
### Cura Version

4.8.0   5.7.0

### Operating System

windows11

### Printer

X5SA 

### Reproduction steps

3DBenchy.stl をDraft 0.2mmでスライス　

### Actual results

天井が逆ピラミッド状になり、船底の文字が消える

### Expected results

スライス結果が変

### Add your .zip and screenshots here ⬇️

[image]

HellAholic, 2024-04-28:
@KouMRT Please include a project file, the issue is most likely from the settings you have selected/modified on your printer. 
Image below is from X5SA sliced with 0.2mm Normal quality (printer does not have a 0.2mm Draft quality)
[image]

GregValiant, 2024-04-28:
@KouMRT - see if "Experimental / Make Overhang Printable" is enabled.  It is a seldom used option that will add chamfers to overhangs to "make them printable".

closed 2024-05-13 by github-actions[bot]