Ultimaker/Cura #18976: supports for models

technodavecairns, 2024-04-28:
### Is your feature request related to a problem?

I've been using cura supports for2 and a half years, its not there anymore, why would I listen to elevator music while my job gets sliced, what are you guys smoking!!  

### Describe the solution you'd like

Bring back the plugin for cylinder supports, it works well instead of support blocker 

### Describe alternatives you've considered

the support blocker leaves unwanted supports in cavities

### Affected users and/or printers

this affects all users and printers

### Additional information & file uploads

how can you think you can print anything with out clean supports

GregValiant, 2024-04-28:
The "Custom Support Cylinder" plugin was written and provided by @5axes .  I don't see it in the MarketPlace for 5.7.0 or 5.7.1 but it is in the MarketPlace for 5.6.0.  I think you can also install if from the 5axes GitHub page.

HellAholic, 2024-04-28:
Hey @technodavecairns ,
It appears you're suffering from a tantrum attack which is unfortunate. 
Nobody is forcing you to install any plugin, so if you do not want something, don't install it.

If you're using Cura for 2.5 years, you should know that it takes time for plugins to be updated after a release, and those plugins are developed by the community contributors who give you "for free" a multitude of different plugins. So show a little respect.

If you would like to support the addition of the plugin to the new version, please send a request to the developer politely and ask them if they can make an update for the new version.

Please note any future comments in the language above will result in your ban from the repository.

closed 2024-04-28 by HellAholic