Ultimaker/Cura #18979: Improving Plugin Longevity in Cura: A Final User’s Perspective and Suggestions

VitorOM, 2024-04-28:
### Is your feature request related to a problem?

Yes, my feature request is related to a problem. The problem is that plugins for the Cura software often disappear from the Marketplace due to the requirement for constant updates. This is frustrating for end users who rely on these plugins for their 3D printing tasks.

### Describe the solution you'd like

 I would like a solution where plugins do not disappear from the Marketplace just because they are not updated. Perhaps Cura could implement a system where older plugins are still accessible and usable, even if they are not updated for the latest version of the software. This would provide stability for end users and ensure that they can continue to use the plugins they rely on.

### Describe alternatives you've considered

An alternative solution could be for Cura to provide better support and resources for plugin developers, to help them keep their plugins updated. This could include more comprehensive documentation, a more user-friendly development environment, or even financial incentives for maintaining plugins. Another alternative could be for Cura to incorporate the most popular and useful plugin features directly into the software, reducing the reliance on third-party plugins.

### Affected users and/or printers

All users.

### Additional information & file uploads

Good afternoon, everyone. I am not a programmer or anything in the field, but I would like to provide feedback/complaint from someone who has been working with 3D printing and the Cura software for over 4 years:

Throughout this period, I have seen various plugins being developed for Cura. As an end user, I have always used many different plugins. However, most of these plugins disappear over time. I remember a few years ago, I used a plugin that managed the nozzle temperature, it disappeared. I also used a plugin that provided very well-calculated preset configurations, it also disappeared…

I contacted a developer and he told me that Cura always requires new updates and if your plugin is not updated, it simply disappears from the list.

This is terrible for us end users, we never have something certain, within a few years it already disappears. I know that Cura is in constant development but this policy that hinders plugins and makes them disappear from the Marketplace is extremely bad. They could very well think about how to improve this aspect.

HellAholic, 2024-04-29:
Hey @VitorOM , thanks for the time you took to submit the request.
I'll take this one on a discussion with the team. Internal reference: CURA-11877

VitorOM, 2024-04-29:
Thank you very much for considering! Even though I'm not an extremely high end user, there are many points to consider that are very relevant to improving Cura.