Ultimaker/Cura #18980: failed to slice a project

bluetriggerfis, 2024-04-28:
### Cura Version

5.0.6

### Operating System

windows 11

### Printer

creality Ender-3 / Ender-3 v2

### Name abnormal settings

i just set the model on a generic ABS profile with some custom settings like walls,infill,support,speed

### Describe model location

its in the middle with the highest ammount of space i can use on this printer

### Describe your model

its a airless basketball modell

### Add your .zip here ⬇️

[image]

bluetriggerfis, 2024-04-28:
it also will not slice on any other profile or other fillament i tested it

GregValiant, 2024-04-28:
Thanks for the report.
That "Airless Basketball" model has been here several times.  It's full of errors.
If you zoom in on the image you supplied you can see red surfaces facing upwards.  All those red surfaces are flipped inside-out and they confuse Cura.
After repairing the model it should slice in 5.7.1.

I'll mark this as a duplicate report and go ahead and close it.  I hope you understand.
Between the model errors and slicing errors in some installations of 5.6.0, this probably can't be made to slice in 5.6.0.

closed 2024-04-28 by GregValiant