Ultimaker/Cura #18982: Gradual lowering bed temperature

Magy357, 2024-04-29:
### Is your feature request related to a problem?

I just wonder about possibility to add feature gradually lowering bed temperature during printing.
As far as I understand, bed heating is important when start printing for best adhesion/stick printing plastic to the bed surface.
But once that first one (or another 2) layer is stuck, the temperature is no longer as crucial as before, so can be lowered.
I understand for some cases turning off heating can cause detach printing build and in that moment whole print would be ruined.
But for example on my printer (Creality Ender) and with stock bed I never had such issue, even when bed is completely cooled down, it require some force and bend bed to detach print.
Couple times I lowered bed temperature manually until "no heating" stage within first 10% of printing and had no issue at all.
So... would it be possible to add such feature for automatic lowering temperature with some sort of setting like (example) decrease temperature for 1°Celsius for each 1% of printing? So from 60°C initial temperature it will decrease by each % of printing down, so the heating will effectively turned off completely when reach room temperature (20-25°C).
That gradual lowering temperature doesn't cause sudden shrink of layer and lowering chance for detach print.
And of course, it can save significant amount of energy.

### Describe the solution you'd like

Feature: Lowering bed temperature
- Start temperature (value)
- Start lowering temperature when print is in x% of progress (value)
- x% of printing progress - end temperature (value)
- End temperature (value)

### Describe alternatives you've considered

no idea about alternatives

### Affected users and/or printers

basically any

### Additional information & file uploads

Very useful for users which want or need to save consumed energy. Especially those who running house/printer on solar panels supply or similar source of energy.

GregValiant, 2024-04-29:
There are a couple of post processors (Extensions / Post Process / Modify Gcode then "Add Script") that you can use for this.
Change At Z is one, and Search and Replace is another.

Change AT Z has a bed temperature setting.  You can have it insert the change at a layer or at a "Z" height.

Search and Replace would be something like:

Search = LAYER:99
Replace = LAYER:99\nM140 S45
Regular Expressions = false

The "\n" is the Python newline character and would be required so the M140 would be on the next line below the layer line.  The "LAYER:99" will change the temperature at preview layer 100.
Either way, you would need a separate instance of the post processor for each change you want to make.
In both cases the "layer numbers" are in base0 and the Cura preview is base1 so you would need to subtract "1" from the preview layer number and enter it in the settings for the post processor.

Magy357, 2024-04-30:
> There are a couple of post processors (Extensions / Post Process / Modify Gcode then "Add Script") that you can use for this. Change At Z is one, and Search and Replace is another.
>
> Change AT Z has [...]

Thanks for that help, but I'm afraid I'm just BFU and such thing you described is like you speaking chinese to me. :D

GregValiant, 2024-05-01:
The tools to do what you want are already in the software.  You need to learn how to use them.  Adding another tool won't help if you can't figure out the ones that are already supplied.

ninjubohra, 2024-07-31:
Hello @Magy357 

It took me a little time to understand what @GregValiant  was describing but I think I found it

After loading your model into Cura

Open the **Extensions** menu item and choose Modify G-code

[image]

You will then see the **Post Processing Script** popup window
[image]

Click the **Add Script** button and choose the **ChangeAtZ** script type
[image]

The settings/configuration options will then presented on the right side
First setting to adjust is the **Trigger** type to be either **Height** (in mm) or **Layer** (a number)
After which you can enable (by checking the box) the property you want to adjust and provide a value
[image]

Note the settings/configuration automatically save (there is no Save Button)

If you want have multiple 'changes' you will need to add the script multiple times with a unique setting/configuration value

Good luck