Ultimaker/Cura #18983: "Use value from material" is missing from the right-click menu on a setting box

GregValiant, 2024-04-29:
### Cura Version

5.7.0 and 571

### Operating System

Windows 10 Pro

### Printer

Ender 3 Pro

### Reproduction steps

1. Made a change to the print temperature.
2. Right-click on the setting box and view the menu options.

### Actual results

1. The right-click menu has two items "remove from favorites" and "configure favorites".  There were 3 options in 5.6.0 and earlier and the top option was "Use value from material".

### Expected results

It could have been removed on purpose, it was useful.

### Add your .zip and screenshots here ⬇️

5.7.0 and 5.7.1
[image]

5.6.0
[image]

HellAholic, 2024-04-29:
Hey @GregValiant ,
I still see the option 🤔 
[image]

Maybe missing a plugin?

GregValiant, 2024-04-29:
I'm going thru it now.  I haven't found it yet, but I do play around with things a bit.  

I had neglected to load "Material Settings".  I guess it comes in with the plugin.

closed 2024-04-29 by GregValiant