Ultimaker/Cura #18985: g-code does not have command for resume after power failure

carolm2011, 2024-04-29:
### Is your feature request related to a problem?

After a power failure my resume button does  nothing and I lose my print

### Describe the solution you'd like

addition of command to g-code to make this possible (M413)

### Describe alternatives you've considered

Don't know what to do except start the print over from scratch

### Affected users and/or printers

anyone who has the misfortune of a power failure

### Additional information & file uploads

My printer is an Anycubic Vyper if that makes a difference

GregValiant, 2024-04-30:
In Cura go to Manage Printers and then Machine Settings.  In the lower left text box is the StartUp G-code.  You can add a line "M413 S1" to the end of the Startup and it will turn Power Loss Recovery on (provided it is enabled in the firmware and dependent on HOW it is enabled in the firmware.)

GregValiant, 2024-05-02:
I'll go ahead and close this.  This is a printer specific feature that would not be added to the main Cura app but rather to the definition file that was submitted by AnyCubic (or an owner/contributor).  It's a simple change that an owner of one of those printers can make.

closed 2024-05-02 by GregValiant