Ultimaker/Cura #18989: 5.7.1 Problem with End Gcode for Modix V4 Big Meter

SeanJaegli, 2024-04-30:
### Cura Version

5.7.1

### Operating System

Windows 10 Home

### Printer

Modix V4 Big Meter

### Reproduction steps

End Gcode set as:
G1 X{move.axes[0].min+2} Y{move.axes[1].max-2} F6000 ; move to the back left
changed in gcode file as:
G1 XNone YNone F6000 ; move to the back left 

### Actual results

results in an error:
GO/G1 insufficient axes homed

printer does not home printhead

### Expected results

printhead should return to home

### Add your .zip and screenshots here ⬇️

Previous version of Cura did not have this problem

GregValiant, 2024-05-01:
Thanks for the report.
Since this seems to have worked previously my thought is that Cura was letting the line pass through unchanged and that the printer was then correctly interpreting the line.
Starting with 5.7.0, Cura now allows math and logic in the StartUp and Ending Gcodes.  I think you have discovered a problem with that with your firmware.
It looks like Cura is trying to decipher your line but it is an illegal statement in Cura as "move.axes[0].min" is undefined and so Cura is evaluating the statement and coming up with "None" as the result in both instances within the line.

(Note that in all these cases the "G1" should really be a "G0" as convention states that a G1 move should be an extrusion and a "G0" a travel move.)

For a printer that has the Origin at the left front corner, a defined bed size of 980 x 1,000 x 1,000 mm, and (for example) a Cura set Travel Speed of 150mm/sec for the print - if the line was to read:
**G1 X2 Y{machine_depth-2} F{speed_travel*60} ; move to the back left**
The Gcode line would read:
G1 X2 Y998.0 F7200 ; move to the back left
which the printer should respond to.

If the printer is an "Origin at Center" machine then the line would be:
**G1 X-{machine_width/2+2} Y{machine_depth/2-2} F{speed_travel*60} ; move to the back left**
And the result in the gcode would be:
G1 X-492.0 Y498.0 F7200 ; move to the back left
which should also be properly processed as a regular move in the printer.

If the printer insists on the form that you show above then you might need to use Search And Replace.
Search = G1 XNone YNone
Replace = G1 X{move.axes[0].min+2} Y{move.axes[1].max-2}

Because this is the way the new version of Cura works - what you are seeing isn't a bug but rather the new "expected behavior".

I'll leave this open so you can respond but I'll remove the bug label for now (it can always go back on).
If you would copy your Ending Gcode from Cura and paste it into a response here it would be helpful.

SeanJaegli, 2024-05-01:
Hi Greg.

Thank you for the quick response. That all makes sense what you explained
to me. We resolved the problem by substituting a fixed integer for both the
X and Y and that of course solved our problem.

Best regards,

Sean Jaegli
<https://globalsurface.ca/> SEAN JAEGLIDIRECTOR
<https://www.facebook.com/GlobalSurfaceSolutionsKelowna/>
<https://www.linkedin.com/in/sean-jaegli-b2ba3139/>
<https://www.houzz.com/professionals/specialty-contractors/global-surface-solutions-pfvwus-pf~1761786847?>
C: 250 215 4737 <2502154737> O: 250 491 3147 <2504913147>
***@***.*** <https://globalsurface.ca/>

GregValiant, 2024-05-02:
I'll go ahead and close this then.

closed 2024-05-02 by GregValiant