Ultimaker/Cura #18997: Slicing does not end

swordfish1945, 2024-05-02:
### Cura Version

5.7.0

### Operating System

Windows 11

### Printer

Sovol SV06

### Reproduction steps

1. Load a model file to build plate.

### Actual results

Slicing starts and never ends. Progress bar cycles back and forth. Slicing will end if I change any setting.

### Expected results

Slicing to end 

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-05-04:
Thanks for the report.
Without a project file we don't have anything to check.  Use the "File / Save Project" command, zip the file, and post the zip folder here.  Posting the Cura.log file ("Help / Show Configuration Folder") would also be helpful.

swordfish1945, 2024-05-04:
Attached please find:Zip file containing the cura.log, the STL file, and a
brief screen recording showing the never ending slicing.

GregValiant, 2024-05-04:
E-mail attachments don't show up here.  You must use a computer to post items.  It's a Git rule.

swordfish1945, 2024-05-04:
I'm not sure what you mean. I included the STL file. Do you mean the CAD
file. I make a lot of my own models. This failure occurs on every model I
try to print with the SV06 printer. I have been using Cura with a Creality
Ender 3 for more than three years.
So, what is this project file. I have always used STL files.

GregValiant, 2024-05-04:
Use the "File | Save Project" command.  It will contain your printer, all your settings, and the model.
Zip that file.
Come to this site and post the zip folder.  Email is OK for a response, but to post a file you must come to the site.

swordfish1945, 2024-05-04:
OK, I've never had occasion to use the Save Project. The Zip is attached.

GregValiant, 2024-05-04:
Email doesn't work.  Drag the zip folder and drop it into the response box.

swordfish1945, 2024-05-05:
[SS_Spacer-LatchSpacer.zip](https://github.com/Ultimaker/Cura/files/15211325/SS_Spacer-LatchSpacer.zip)

swordfish1945, 2024-05-05:
Here is zip
[cura.zip](https://github.com/Ultimaker/Cura/files/15211344/cura.zip)

swordfish1945, 2024-05-05:
> Email doesn't work. Drag the zip folder and drop it into the response box.

[cura.zip](https://github.com/Ultimaker/Cura/files/15211349/cura.zip)

GregValiant, 2024-05-05:
OK, I got it.
I didn't have any trouble slicing the project.  Looking at the log file, there are a lot of errors.  There are references to the new plugin "CuraEngine Tiled Infill Generation".  I suggest you go into the Cura Marketplace and disable that plugin.

For the rest someone from the Cura team will need to take a look.

swordfish1945, 2024-05-05:
I have removed the plugin but did not fix the slicing issue. I looked at the log last week and noticed errors, but they don't mean much tome.
Thanks for the help.