Ultimaker/Cura #18998: bad result of slicing, flash of all detail in preview

SurfBlaster, 2024-05-02:
### Cura Version

5.7.0

### Operating System

W10x64 LTSC

### Printer

Generic, Prusa i3 compatible

### Reproduction steps

Place parts on the table for sequential printing. Group them for sequential printing. Slice. On the 2nd or 3rd layer of the first part we are guaranteed to get a problem.
[image]
[image]
[image]
You see, in the preview the entire part is highlighted, although there should only be one level.

### Actual results

At first I thought that this was just a defect in the preview, but no, on the print in this place there is some kind of layer of the wrong height, and the part falls apart.

### Expected results

Perhaps the variable height layers are not working correctly somehow. However, this does not happen in version 5.6.0; I ended up going back to this old version.

### Add your .zip and screenshots here ⬇️

[err-test-proj.zip](https://github.com/Ultimaker/Cura/files/15185845/err-test-proj.zip)
Here is my project causing slicing error, hope this will help you find the error.

GregValiant, 2024-05-02:
Thanks for the report.
If you turn off "Adaptive Layers" the problem goes away.  You don't need "Bridge Settings" for this project either.

The Cura team will take a look.

GregValiant, 2024-05-03:
I left the bug label on this because I understand that for some projects turning off Adaptive Layers would not be an option.
The project you posted didn't need Adaptive Layers (or Bridge Settings) so I mentioned the workarounds.
If you can't post the exact project file (proprietary models or whatever) I understand but let us know that you posted a "representative" project rather than the "real" project..