Ultimaker/Cura #18999: Sublevels: quick alternative to adaptive layer height

SurfBlaster, 2024-05-02:
### Is your feature request related to a problem?

Many criticize this slicer for its inability to control adaptive layers, and this really makes you waste a lot of extra time when printing where it is not really needed. But even with full control of adaptive layers, we still spend extra time on detailing the monolithic parts of the part...

### Describe the solution you'd like

It seems to me that it is possible to introduce “refinement” layers at the boundaries of the model. Only these areas need to be printed in “thin” layers. A simple solution is to enter a refinement divisor, for example a number in the range 2...5. The base layer height will be divided by this number and additional layers will be printed individually, starting with the thinnest one and ending with the "base layer".

Here, in the picture, I roughly outlined the order of printing such layers.
[image]

### Describe alternatives you've considered

If you go even further, you could make crude non-planar printing using a similar principle. What I mean is that you don't need to do full-fledged non-planar printing, it requires special printer mechanics. But within one base level, the nozzle can smoothly lower and rise, something like what Arachne does for line width, here, the same could be for height (within one base level). A possible problem is that the layers are too thin, well, you will have to cut them off, probably some minimum thickness should be the limit.

### Affected users and/or printers

Such an option could even be enabled by default. That is, this can concern literally everyone. To get an accurate model on print in just a little more time than a rough model - what could be better?

GregValiant, 2024-05-05:
I'm trying to get my head around this.  It looks to be really, really complicated.

- In your image, a layer is split into 6 sub-layers.
- Each sub-layer has a different line width.
- Sub-layer "1" is a lot taller than it is wide.  That isn't likely to happen in real life.
- Sub-layer "2" is much wider than the rest of them.  It would seem to require massive over-extrusion.
- All of the sub-layers seem to encompass just the "wall" area of a print so that all inner and outer walls are affected but things like supports, infill, prime towers, ooze shields, and skins are still the original layer height?

SurfBlaster, 2024-05-06:
This is a very rough picture, just to illustrate the idea. In real life, naturally, any of the drawn polygons will have a width much greater than the height.

__
Offtopic:

However, I tried some experiments, leaving gaps in the model, and then trying to fill these holes with a massive injection of plastic (“fill the hole two or three layers deep”), but it turned out very poorly.

Unfortunately, I can’t carry on a conversation on this topic because I don’t understand how the slicer works. If I were coding a slicer, perhaps the internal representation of the model would be something like voxels, and within such a model such inconsistencies could be resolved; however, judging by the errors in modern slicers, there is clearly something else going on there. Sorry!