Ultimaker/Cura #19002: Bad slicing result, unexpected gaps

stefanino-ch, 2024-05-02:
### Cura Version

5.7.0

### Operating System

Windows 11

### Printer

Sovol SV04

### Reproduction steps

Add part to cura. 
Part files are attached. 

Slice it. 

### Actual results

Result in v5.4
[image]

Result in v5.7
[image]

Happens with .stl and with .3mf file. 
The .stl file I have checked with Blender, perfect display, no errors in model reported. 

I have not even tried to print, as the gaps displayed are so big. 

### Expected results

At least the same result as in previous versions. 

### Add your .zip and screenshots here ⬇️

Part files
[gear-wheel-big.zip](https://github.com/Ultimaker/Cura/files/15189675/gear-wheel-big.zip)

Universal project file
[UCP_gear-wheel-big.zip](https://github.com/Ultimaker/Cura/files/15189704/UCP_gear-wheel-big.zip)

GregValiant, 2024-05-02:
Thanks for the report.
Neither 3mf file opened as a project for me, but what I see in your image is exactly the preview to expect when "Coasting" is enabled.
The preview only shows moves with extrusions.  Because "Coasting" moves aren't extrusion moves, they aren't shown and so you see the characteristic gap.  The nozzle moves across as if it was still an extrusion, but the extruder doesn't turn and the pressure within the nozzle is allowed to bleed off.

I'll go ahead and remove the bug label and close this.  Let me know if you feel that is premature.

closed 2024-05-02 by GregValiant

stefanino-ch, 2024-05-05:
Hey Greg
You're right. I've disabled coasting and get the expected result. 
[image]

PIBCAK: Problem Is Between Chair And Keyboard. :-(

Thank you for your support.

GregValiant, 2024-05-05:
Coasting works as it should.  Just remember that the preview will look a little odd.