Ultimaker/Cura #19005: Interleaved Prime Tower printed after Model

jffmichi, 2024-05-02:
### Cura Version

5.7.1

### Operating System

Arch Linux

### Printer

Custom

### Reproduction steps

Generate a model that extruder 1 on layers 1 to 19 and extruder 2 and 3 on layer 20
Select "Interleaved" as Prime Tower Type (default value)

### Actual results

On the topmost layer (20) the print order is as follows:

Model (Extruder 2)
Prime (Extruder 2)
Prime (Extruder 3)
Model (Extruder 3)

### Expected results

The print order should be as follows:

Prime (Extruder 2)
Model (Extruder 2)
Prime (Extruder 3)
Model (Extruder 3)

### Add your .zip and screenshots here ⬇️

[image]
[prime_tower_after_model.zip](https://github.com/Ultimaker/Cura/files/15191588/prime_tower_after_model.zip)

jffmichi, 2024-05-07:
@GregValiant you added the "Needs Info" status. What information do you need and how can I help?

GregValiant, 2024-05-07:
Sorry, I must have left the page without commenting.

I need a regular project file ("File | Save Project").  The UCP file simply wants to use my printer and that won't work.

jffmichi, 2024-05-07:
Like so?

[prime_tower_after_model.zip](https://github.com/Ultimaker/Cura/files/15239084/prime_tower_after_model.zip)

GregValiant, 2024-05-07:
This looks like a bug in the "Interleaved" prime tower.  When I switch to "Normal" it appears that all the tools hit the prime tower as they should.  Make that change and I think you'll see that it is correct.

The Cura team will take a look.  This isn't the first problem that has come up with the "Interleaved" feature, (it's new) but the other one had to do with the brim-to-tower interface.

@HellAholic  - I can reproduce this with the project file and with my own printer (configured with 3 extruders) and with an S5 (alter the model to eliminate the 3rd extruder).

jffmichi, 2024-05-07:
Switching to "Normal" prime tower does fix the order on the topmost layer. However, it generates filament switches on EVERY layer, even if the model only uses extruder 1 on those layers. Since I'm switching the filament manually this is not really a feasable workaround.

GregValiant, 2024-05-08:
I understand the problem.  The prime tower is treated like another model so each extruder must keep up it's level in pace with the print.  That works well when everything is automated, but not so good when doing manual changes.  I would assume that the Interleaved prime tower should actually work like the Normal prime tower.  That's actually where I see the bug.

If you don't use the purge tower you would get the pause with your M0, but the nozzle would be on the print so you couldn't purge by hand.  I ran into the problem you're having and ended up writing a post-processor that does what you are trying to do.  It goes through the gcode and replaces the tool changes with code that retracts, parks, unloads, pauses, reloads, purges, retracts, and moves back to print.  It's still a PITA with all the pauses.  It also has a set of rules because Cura needs to be set up "just so" in order for it to work.  I can use it, but it's more of a proof-of-concept thing than a "for everybody" finished product.

jffmichi, 2024-05-08:
Are you saying the interleaved prime tower should also change filament on every layer? While I agree that this would probably work with an automated system it seems like a waste of time to insert 3 extruder switches on every layer when 2 of the extruders are not used.

(On my machine M0 also moves up a little. I can purge manually but a purge tower with a fast move back to the actual model gives better results, less time for filament to ooze out)

The interleaved prime tower seems like the perfect solution, it doesn't cause any unnecessary extruder switches and also provides support for the upper layers.

HellAholic, 2024-05-08:
Added a ticket for discussion. Internal Cura reference: CURA-11900

@GregValiant , I couldn't get it to misbehave on the S5 so if you could share the project saved on that one it would be great!

GregValiant, 2024-05-08:
@HellAholic I believe that's a mistake on my part.  On the S5 the nozzle isn't going to the prime tower because it doesn't need to.  

Here is a project with my printer configured with 3 extruders.  The sequence is:
T0 finishes layer 20
T1 starts printing on layer 21 without a visit to the prime tower.
T1 goes to the prime tower and primes
T2 is at the prime tower, primes, and prints on layer 21
print ends

I know the OP is tr(ying to configure a single extruder printer for manual changes, but the problem seems independent of that.  On the top layer - T1 is priming AFTER it is used.
(It makes me wonder if this might be specific to printers configured with more than 2 extruders.)

[GV_prime_tower_after_model Ender3Pro.zip](https://github.com/Ultimaker/Cura/files/15247275/GV_prime_tower_after_model.Ender3Pro.zip)