Ultimaker/Cura #19012: cura takes too long to load stl

juanca-21, 2024-05-03:
### Cura Version

5.7.1

### Operating System

mac os Monterey

### Printer

creality ender 3

### Reproduction steps

try to load an stl file, any

### Actual results

it takes more than 10 minutes to show it

### Expected results

to load faster

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/15202367/cura.log)

GregValiant, 2024-05-04:
Thanks for the report.
You have a bluetooth or other serial devices connected to your computer?  Cura is trying to connect.  Go to the Marketplace and click on the "gear" icon.  Scroll down and disable "USB Printing".
Without a project file there is no way to investigate this.  The instructions are on the form you filled out.

juanca-21, 2024-05-04:
Many thanks. I will try your sugestion. 
i didn’t upload a proyect becouse i thought it wasn’t necessary. 
please accept my apology

closed 2024-05-04 by juanca-21