Ultimaker/Cura #19016: can't change sitting

Rushbee1, 2024-05-04:
### Cura Version

5.7.0

### Operating System

window 11

### Printer

Elegoo Neptune 4

### Reproduction steps

change sitting 

### Actual results

slicing does not stop

### Expected results

put sitting back  

### Add your .zip and screenshots here ⬇️

[EN4_Stephany Key Ring Hook v1.zip](https://github.com/Ultimaker/Cura/files/15210951/EN4_Stephany.Key.Ring.Hook.v1.zip)

GregValiant, 2024-05-04:
Thanks for the report.  The project would not slice for me until I moved the model away from the mid point of the build plate

@HellAholic you may want to review this one.  I used 5.7.1.  Tiled Infill Generator?

HellAholic, 2024-05-05:
@GregValiant could you share your log files please? For me it slices in the center without issue (personal PC). I'm wondering if this is related to svg write permission (admin bug as we know it) or something else. I'll check it on another device later to be sure.

GregValiant, 2024-05-05:
@HellAholic - I can't duplicate it now.  The project asks me to load the Tiled Infill Generator which I decline (as I did the first time) and since that first crash, it always slices OK now.  I've tried with 5.7.0 and 5.7.1.
My log has refreshed and so the crash isn't in there.

HellAholic, 2024-05-08:
On one device (PC) when I follow your path of (Open anyway > Slice) it generates an error and on another one, it just slices normally, other times it slices without infill. But that is why we have warnings in place for trying to slice a project that has plugin dependency.

Outside of that, there are 500+ settings that are connected to extruder number as an example, if you switch the extruder number, then a lot of those settings need to be recalculated, pressing the slice button in that period of time (that can be couple of seconds depending on the device's performance) could lead to slice crashes.

We have a couple of tickets in the oven to improve the performance for the settings, so hopefully these type of issues will be less noticeable in near future.

Lastly, we have the sentry in place so if there is an issue that we cannot reproduce by following the steps, it will be caught there.