Ultimaker/Cura #19017: Slicing Failed when 'Z-Hop Only Over Printed Parts' enabled

gh13, 2024-05-05:
### Cura Version

5.7.1

### Operating System

Windows 10

### Printer

Ender 3 S1 PRO

### Name abnormal settings

Z-Hop Only Over Printed Parts Enabled

### Describe model location

Centred on build plate. Won't slice if moved either

### Describe your model

Model is watertight, no errors, Can slice same model without enabling Z-Hop Only Over Printed Parts

### Add your .zip here ⬇️

[Failed Slice Project.zip](https://github.com/Ultimaker/Cura/files/15211884/Failed.Slice.Project.zip)

GregValiant, 2024-05-05:
Thanks for the report.
I wasn't able to duplicate the problem on my Win10 Pro laptop.  If you would post your log file ("Help | Show Configuration Folder") it may point to your problem.

gh13, 2024-05-05:
Here you go @GregValiant 

[cura log.zip](https://github.com/Ultimaker/Cura/files/15213057/cura.log.zip)

GregValiant, 2024-05-05:
Thanks for that.
From the log file:
2024-05-05 14:03:16,098 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2024-05-05 14:03:16.098] [error] **The file %s could not be opened for writing.**
2024-05-05 14:03:17,955 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [1038]: **Backend exited abnormally with return code 3221226505!**
2024-05-05 14:03:17,959 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [421]: Attempting to kill the engine process

The auto-reporting function seems to have worked so UM should get the rest of the crash info.

gh13, 2024-05-05:
Thanks for the update @GregValiant. I saw the error in the file but to me it may as well have been written in Egyptian so hoping the devs can decipher it

HellAholic, 2024-05-05:
Hey @gh13 , quick workaround for this is to run Cura as administrator.
We have a ticket to address this issue (bug has been arrested). Should be resolved in the next release.
I'll mark this one as duplicate (relates to SVG).

gh13, 2024-05-05:
Thanks @HellAholic I can confirm that when Cura is Run as Administrator the project slices successfully when Z-Hop Only Over Printed Parts Enabled. Good to know it's a known bug and not something silly I was doing.
There were a couple of errors in the log which weren't in the original log I uploaded. I've attached the latest log along with the debug.html referenced if that has any sort of significance.

[cura log 2.zip](https://github.com/Ultimaker/Cura/files/15214031/cura.log.2.zip)
[debug.zip](https://github.com/Ultimaker/Cura/files/15214033/debug.zip)