Ultimaker/Cura #19019: Bed Temperature Wrong with Snapmaker Artisan

AlexanderWMacFarlaneIV, 2024-05-05:
### Cura Version

5.7.1

### Operating System

Linux 

### Printer

Snapmaker Artisan

### Reproduction steps

Create 2 custom materials and set the build plate temperature to 105 in both.

Select snapmaker Artisan as the printer after installing the plugin from the marketplace.

Change both build plate temperature and initial build plate temperature to 105 for both materials in the print settings,

Slice.

Run Gcode on a Snapmaker Artisan printer

### Actual results

The build plate temperature only goes up to 80.

### Expected results

The build plate temperature should be 105

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/15213569/cura.log)

AlexanderWMacFarlaneIV, 2024-05-05:
[Noble_Woman_Luban.txt](https://github.com/Ultimaker/Cura/files/15213836/Noble_Woman_Luban.txt)
[Noble_Woman_Cura.txt](https://github.com/Ultimaker/Cura/files/15213837/Noble_Woman_Cura.txt)

I have narrowed down the problem to something in the Gcode header.
Attached are 2 gcode files renamed to .txt so I can post them here.

One is generated by Luban and the other by Cura,

I was able to solve the problem by pasting the entire header from the Luban file to the Cura file, so there is something that Cura is not doing right with that header which causes the build plate temperature to be wrong.

HellAholic, 2024-05-05:
Hey @AlexanderWMacFarlaneIV ,
Please raise this with the printer manufacturer and the plugin developer.
You are using a plugin for the printer that is not reviewed by Cura team.
The header you mentioned is not something that is generated by Cura directly but rather part of the plugin for the printer.
In the files you shared, the header section is generated by the plugin. What is generated by Cura is from line 28.
`;FLAVOR:Marlin
;TIME:4181
;Filament used: 3.2132m, 1.65295m
;Layer height: 0.16
;MINX:76.064
;MINY:113.345
;MINZ:0.336
;MAXX:178.823
;MAXY:184.571
;MAXZ:22.286
;TARGET_MACHINE.NAME:Snapmaker Artisan
;Generated with Cura_SteamEngine 5.7.1`

closed 2024-05-05 by HellAholic

AlexanderWMacFarlaneIV, 2024-05-05:
I guess I can repost it somewhere but do you see what the problem is?

HellAholic, 2024-05-05:
I can't make any judgment to point out the issue without the knowledge of the printer.
But only comparing the two Gcodes, the format that needs to be prepared for the printer needs an index for the buildplate P0
So both M140 and M190 need to have the P0 appended before the S105.
If that resolves the issue, then you can either raise it with the plugin developer to add a post processing step, or you can add a post processing script from Cura yourself to search and replace the commands.
[image]

However, if the issue is within the lines above (header section line 1-40 in the Luban) then the steps above won't resolve the issue as it would be the line 26 of Luban which is supposed to be a comment but the firmware is treating it as instructions.
[image]

You can also modify this as a workaround with the same post processing script.

AlexanderWMacFarlaneIV, 2024-05-05:
Ok Thank you.

I will look at your suggestion and post any other question on the plugin authors Github page.