Ultimaker/Cura #19020: When Cura first started, models slow to load.

hammer-fired, 2024-05-05:
### Cura Version

5.4 - 5.71

### Operating System

Fedora Linux

### Printer

Creality Ender 3 S1 Pro

### Reproduction steps

Start Cura
Select model to load

### Actual results

Simple model takes about 45 seconds to load. Afterwards, models load instantly. If I wait a minute or so after Cura is started, then load a model, it loads instantly, so it appears that Cura is still doing something in the background even though it appears to be fully started.

### Expected results

Model should load instantly as in previous versions.

### Add your .zip and screenshots here ⬇️

Nothing to see.....

cpu012, 2024-05-10:
I can confirm that I have the same issue on fedora linux. it happens in the flatpak and in the rpm version

hammer-fired, 2024-05-11:
I should have noted that 5.4 is from a RPM and 5.7.1 is an appimage.

cpu012, 2024-05-13:
I found a workaround
[https://github.com/Ultimaker/Cura/issues/18794](https://github.com/Ultimaker/Cura/issues/18794)

hammer-fired, 2024-05-13:
Bingo! That fixed it!