Ultimaker/Cura #19026: select face to align to the build plate bug

trixaras, 2024-05-06:
### Cura Version

5.7

### Operating System

win11

### Printer

ender3

### Reproduction steps

There is a bug that wasnt there on previous versions. select face to align to the build plate does not work as it is supposed to.

### Actual results

no use

### Expected results

it should align a models surface to the buildplate

### Add your .zip and screenshots here ⬇️

[screeen1.zip](https://github.com/Ultimaker/Cura/files/15220403/screeen1.zip)

GregValiant, 2024-05-06:
Thanks for the report.
I can't duplicate this as it works fine on my system.  Post a project file and the Cura.log file.  The instructions are on the page you filled out for this report.  Al the screen shot shows is your model.

ansonl, 2024-05-28:
@GregValiant  I found the same alignment issue happening on one of my models. The model just ends up rotated at different weird angles depending on where you click to align. 
I can't publicly share the specific model that it happens on. The problem goes away when I cut the model in half and try to align a cut version. Is there a way to send the model to a member of the Ultimaker team directly for reproduction? It's a small enough 3MF that is under 2mb.

widsinga, 2024-07-15:
I have the same problem since I installed 5.7.1. Different models and different OS (Windows/MacOS) face alignment is definitely broken.