Ultimaker/Cura #19031: "Reload All Models" only reloads one model

gsangermano, 2024-05-06:
### Cura Version

5.7.1

### Operating System

Windows 11

### Printer

Creality Ender 3 Pro

### Reproduction steps

1. Load a model
2. Right click on the model and select "Multiply Selected" and make 3 copies
3. In another program, save an updated version of the  model. Or in the file system, delete the model and rename another to have the same name.
5. In Cura, from the "File" menu select "Reload All Models"

### Actual results

Only 1 model is updated.

### Expected results

All 4 models are updated.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-05-06:
Thanks for the report.
It appears (to me) that when you multiply the model and Cura assigns a unique name to each copy by appending "(nr)" to each name, that only the model with the original name ("x.stl" in your project) is acknowledged as being the same model.  So x.stl(1), x.stl(2), etc. don't get altered in the Cura scene because of the different name.  I think that makes sense, but whether or not that is actually what's happening might be incorrect.

Someone from the Cura team will take a look.

HellAholic, 2024-05-09:
When you multiply or duplicate a model, it creates a copy at that exact moment with all the per model settings. However, the newly generated copies don't have any reference to the original STL that was opened, so changing the original STL will not affect them when reloading all models since they only inherited from the original at the moment of being copied.
I understand what you're trying to do here but using the multiply as a template system is not part of the multiply function so
I suggest changing this to a feature request as that seems more in line.

gsangermano, 2024-05-10:
What you guys are saying makes sense except for one thing: this used to work. I've done this literally dozens of times in the past.

HellAholic, 2024-05-10:
@gsangermano I'll add a ticket for investigation.

katyo, 2024-09-20:
I have same issue with 5.8.1

GregValiant, 2024-09-20:
This did work as described (all models updated) in 4.13.1.  Checking 5.8.0 only the original model updates.
4.13.1
[image]

5.8.0 Only the original model (in the middle) updates.
[image]

GregValiant, 2024-09-20:
Well messy sucks as it often leads to what we already have here and that is an _"Unexpected Consequence"_.

Updating all the models is important to professional designers/technicians.
They lay out their build plate with multiple copies of a model, look at the slice, and realize that if they make a minor tweak to the model the slice will result in a better print, less supports required, better orientation ... whatever.
So they reload the model and only the initial model updates.  Unfortunately, they have added Support Blockers to all the models and also unfortunately, those blockers are linked to the individual models which now need to be deleted.  They have to re-do the entire build plate (except for the original) for what was a minor change to the model.  That's a serious hit time-wise, and time is money.  A real downside is that if it is a very minor change to the model, and it is internal, that some users may not know that the copies did not update.  That's a bad situation.

Whether it's messy or not, I would advise to bump this up.  I know a lot of development is driven by the kids printing mini's and dnd characters.  The high-end users need some love sometimes too.  They are the ones most likely to be UM customers, or future UM customers.

yet-another-average-joe, 2024-10-12:
Some news about this major bug ?
It makes Cura unusable when iterating and printing series of pieces.

GregValiant, 2024-11-16:
@HellAholic, 
These bug reports seem to be related to this.  Can these also be closed?
[#18421](https://github.com/Ultimaker/Cura/issues/18421)
[#11557](https://github.com/Ultimaker/Cura/issues/11557)
[#10137](https://github.com/Ultimaker/Cura/issues/10137)
[#15641](https://github.com/Ultimaker/Cura/issues/15641)

gsangermano, 2024-11-22:
Bad news. I just downloaded the 5.9.0 and this bug is NOT fixed (actually it's half fixed). 

If you look at step 3 in my original steps to reproduce, the first case works. That's were you save a new version of your file into the old version.  But not the second case where you delete the file, rename a different file to the same name and do File \ Reload All Models. Only the original model is loaded with the new one.