Ultimaker/Cura #19034: slice fail

ArashiKingu, 2024-05-07:
### Cura Version

Elegoo Cura 5.6.0

### Operating System

Windows 11

### Printer

Elegoo Neptune 4 plus

### Describe model location

center of plate

### Describe your model

flexi Raptor. its not water tight. when using normal cura it slices fine. it fails during elegoo cura

### Add your .zip here ⬇️

n/a

uNbAs, 2024-05-07:
I have a same situation, but system is different.

> Cura Version
5.6.0 and 5.7.1 (appImage)

Operating System
Ubuntu 22.04

Printer
Creality Ender 3

Name abnormal settings
No response

Describe model location
center of plate

Describe your model
Calibration cube

GregValiant, 2024-05-07:
@ArashiKingu ,
If the problem is with Elegoo Cura then the bug report should be made to Elegoo.  This is the "UltiMaker Cura" GitHub and has nothing to do with other forks of Cura (Creality, Elegoo, Longer, etc.).
I'll go ahead and remove the bug label and close this.  I hope you understand.

closed 2024-05-07 by GregValiant