Ultimaker/Cura #19035: Failed Slice

WernerHerr, 2024-05-07:
### Cura Version

5.7.0

### Operating System

Windows 10

### Printer

Prusa I3

### Name abnormal settings

None

### Describe model location

Slightly forward

### Describe your model

Bracket

### Add your .zip here ⬇️

[Tumbler Motor Face Bracket v9.zip](https://github.com/Ultimaker/Cura/files/15232772/Tumbler.Motor.Face.Bracket.v9.zip)

GregValiant, 2024-05-07:
Thanks for the report.
I can't duplicate the problem.  The project slices fine for me and I didn't find any errors in the model.

Could you post the "Cura.log" file?  You can find it using "Help | Show Configuration Folder".
Someone from the Cura team will take a look.

As an aside (and this is purely opinion) - If this is a functional piece (as opposed to a mock-up part) I'd print it like this with extra walls around the counterbored holes and less dense infill but with an infill wall count of 3:

[GV_Tumbler Motor Face Bracket v9.zip](https://github.com/Ultimaker/Cura/files/15234327/GV_Tumbler.Motor.Face.Bracket.v9.zip)

[image]

WernerHerr, 2024-05-07:
Hi
Thank you so much for the reply
I rolled back to 5.5.0, and it worked fine. There was probably a problem during the install
I really appreciate the feedback

Regards
Werner

Sent with [Proton Mail](https://proton.me/) secure email.