Ultimaker/Cura #19050: Adventurer 3 starts printing mid-air

rmeissn, 2024-05-10:
### Cura Version

5.7.1

### Operating System

Fedora 40

### Printer

Adventurer 3

### Reproduction steps

1. Install Adventurer 3 from the recently added profile
2. Slice a model
3. load to the printer via usb stick
4. start printing

### Actual results

The printer heats up and starts printing slowly (as expected) in mid air (half the printers available height), not at the print bed itself.

### Expected results

The printer starts printing at the print bed.

### Add your .zip and screenshots here ⬇️

[A3_TrueFruits750.zip](https://github.com/Ultimaker/Cura/files/15274215/A3_TrueFruits750.zip)

rmeissn, 2024-05-10:
Just for the record, I've previously set up my own custom print profile for the printer, which works as expected, using these start/end gcodes:

```
# start gcode
M140 S{material_bed_temperature} T0
M104 S{material_print_temperature} T0
M104 S0 T1 ; Tool 1 to 0°C
M107 ; Fan off
M900 K0.000 T0 ; K-Factor to 0
G90 ; Absolute Positioning
G28 ; Home XYZ
M132 X Y Z A B ; Recall home offsets
G1 Z50.250 F420 ; Move Z to 50mm
G161 X Y F3300 ; Move Axes to Minimum
M7 T0 ; Stabilize bed temperature
M6 T0 ; Stabilize extruder temperature
M651 S255 ; Set case fan on
M108 T0 ; Break and Continue

# end gcode
M104 S0 T0 ; Hotend to 0°C
M140 S0 T0 ; Bed to 0°C
G162 Z F1800 ; Move Z Axes to Maximum
G28 X Y ; Home XY
M132 X Y A B ; Recall home offsets
M652 ; ???
G91 ; Relative Positioning
M18 ; Disable Steppers
M107 ;Turn-off fan
```

rmeissn, 2024-05-10:
Well, I guess you're right, that flashforge wants people to use their slicer (or just doesn't care about standards).
Nevertheless, I do see these cura/flashforge-integration problems here:
- The cura-included profile for the Adventurer 3 produces valid gcode, that won't print correctly on/match the machine
- The things I apply are pretty basic, (exchange G0 with G1, split Z coordinate commands, convert temps from float to int) -> cura should be able to support these and just activate these things/scripts, when the Adventurer 3 profile is used

Don't get me wrong: The things I apply with my script are leading away from industry standard usage of gcode. But the goal (at least my goal) is to use cura for the machine, not to exclude it because some simple fixes need to be applied.

If cura isn't providing a "if this printer is selected, automatically setup these post processing scripts" (e.g. a python or shell script), then it should be supported in the future to add further printers to cura -> feature request
Alternatively, there are already gcode flavours supported by cura and this seems to be a flashforge gcode flavour to me -> feature request for flashforge gcode flavour

GregValiant, 2024-05-10:
I don't work for Ultimaker. I'm a "member of the community" and have made a couple of minor contributions to Cura.

In order to do what you need, someone has to have a Flash Forge printer.  Ultimaker isn't about to go out and buy one of every printer on the planet to check them and incorporate them into UM Cura.  That leaves the integration to either Flash Forge (unlikely) or an owner like yourself.
I dabble in post-processors and so far as I've read, what is apparently required to convert the standard Cura gcode to something that would be acceptable to your printer seems to be something that can be done in post-process (or with an "Extension" (plugin)).
I don't know your printer and without one to debug with, this is something I simply can't do and neither can UltiMaker.
Does Flash Forge bother to produce a listing of their supported G and M codes, the definitions and parameters, etc.?  That would seem to be a requirement for a competent post-processor/plugin.

GregValiant, 2024-05-10:
This is a post-processor for Cura.  To install it - unzip and copy "CuraToFlashForge.py" to the "scripts" folder in your configuration folder.
"Convert Cura to Flash Forge" will be available under "Extensions" with the other post processors.

The script will:
- Convert all G0 moves to G1 moves.
- Split any Z moves to their own lines.
- Add a Z offset to all Z values if required.  Setting it to 0.0 will disable.
- Convert any floats in M104, M109, M140, M190 and M106 to integers (rounds up).
It incorporates what I saw on you Git page.  Let me know if there are issues.  This was quick and dirty.
[CuraToFlashForge.zip](https://github.com/Ultimaker/Cura/files/15276966/CuraToFlashForge.zip)

rmeissn, 2024-05-11:
Many thanks for the script! This seems much simpler to use than my bash script.

Yeah, I guess it can't spiralize.

I've tested the script (with a larger model, Cura 5.7.1), but got:
- I still see `G0` commands within the gcode and Mixed `G0 X Y Z` commands
- M140 in the start gcode is not processed - still e.g `M140 S55.0 T0`, not `M140 S55 T0`

I'm currently not familiar enough with cura python post-processing scripts to resolve this myself. Maybe when I got some time left.

GregValiant, 2024-05-11:
Post that gcode.  The script is short and processes every line in the gcode.  If it ran (did not error out) then it should not have left any remnants.
The script must be the last script in the list of post processors.  If other posts add lines after that script has run, then they would not be affected.

GregValiant, 2024-05-11:
This is the same script.  I've added a "Log" entry that will show up in the Cura.log file as:
"INFO - [MainThread] PostProcessingPlugin.PostProcessingPlugin.CuraToFlashForge.execute [64]: Convert Cura to Flash Forge started"
When the post processor runs successfully this message will appear:
[image]

This is the new one.  Over-write the other version in the "scripts" folder and enable it using "Extentions/Post Processing/Modify Gcode" and "Add Script".  If there are other post-processors running - this one needs to be last in the list.
[CuraToFlashForge.zip](https://github.com/Ultimaker/Cura/files/15282860/CuraToFlashForge.zip)

rmeissn, 2024-05-12:
Many thanks for the improvements! I've tested the script and it is executed, but didn't work well as the printer ignored the saved bed mesh. I had to alter the script to respect the bed mesh and edited a few minor things to get closer to the gcode produced by Flashprint: 
[CuraToFlashForge.zip](https://github.com/Ultimaker/Cura/files/15287146/CuraToFlashForge.zip)

The script is tested with my Adventurer 3, running the latest available Flashforge firmware. The printer prints fine with the file named `*.gcode`. I can't tell about other printers and firmware versions.

Maybe this script can be added to the Adventurer 3 profile, already in Cura, in order to get it working.

GregValiant, 2024-05-14:
The author of the Flash Forge plugin could incorporate it into that I guess, or it could be a regular post-processing plugin as it currently is.  I'm not at all qualified to take on a QML/Extension project to get the commands into the Cura front end.
As a post-processor it's short and to the point.  The question really is what other Flash Forge printers and Flash Forge firmware it would be good for.  Right now the title says Cura To Flash Forge, but is that true?  Maybe it would be better as Cura To Adventure3 as that is the only machine it's been checked on.  Some feedback from Flash Forge would be a good thing.

rmeissn, 2024-05-15:
@Jeremie-C Can you have a look and comment on this topic? You said in https://github.com/Ultimaker/Cura/pull/15613 that your profile has been tested on an Adv3, but mine refuses to work with your setup, needing the above posted post-processing script.