Ultimaker/Cura #19051: Slicing failing with supports

Denied-X, 2024-05-10:
### Cura Version

5.7.0

### Operating System

Windows 11

### Printer

Ender 3 v2

### Name abnormal settings

Supports

### Describe model location

no

### Describe your model

My model is a simple and small flower pot

### Add your .zip here ⬇️

[Project_material.zip](https://github.com/Ultimaker/Cura/files/15278395/Project_material.zip)

GregValiant, 2024-05-14:
Thanks for the report.
Slicing with both "Normal" supports and with "Tree" supports the model sliced for me without a problem.
[image]

So on my Win 10 Pro system and with both Cura 5.7.0 and Cura 5.7.1 I'm unable to duplicate the problem.
If you could post the "Cura.log" file it would be helpful.
The Cura team will take a look.

If you have that model, putting a 45° chamfer on that outside lip (that requires support) would eliminate the need for support.  I know that's not very helpful here, but doing that would cut your material usage way down.  A lip that is only 0.50 mm wide is very difficult to support whether you use "XY overrides Z" or "Z overrides XY".
[image]

Denied-X, 2024-05-14:
Thank you so much for your help! The thing is I was trying to print it upside down, I'm still new in this field and I'm testing a lot of things out to try and learn from the outcomes. I've still never used supports till now so I wanted to try with a simple design I have and I got the error then saw that I can report the bug. So I did, but thank you at least now I know its working so I can figure it out further. Have a nice day!

GregValiant, 2024-05-15:
On that particular model - if you print it upside down the inside will need to be full of supports to hold up the roof (the bottom of the pot).
There is a rarely used setting in Cura called "make overhangs printable" that will add a chamfer to that outside edge so it won't need support.  Remember to turn it off when you are done because it can cause some odd looking issues on models that don't need it.

I do see something about the slice that I don't like though.
@HellAholic, When I slice this model with a 0.40 line width the walls jump from a single wide extrusion, to side-by-side narrow extrusions causing wavers in the wall in the preview and I believe they would up up in the print.  Just for a test I sliced it with a 1.00 line width and the walls look as they should.

Both of these are with the line width at 0.40 in 5.7.1.  I didn't find any variations in the thickness of the cross-section of the wall which is 0.82 thick.
[image]

[image]

GregValiant, 2024-05-15:
Thanks for looking.
Hole Horizontal Expansion at 0.10 with the Max Hole Diameter at 0.0 caused the problem.

The model on the right has a "Per Model" Max Hole Diameter of 8.0.  
[image]

When you get this project open also take a look at the Z-seam with "Z-seam Relative" on and then off.
[GV pot_draft1.zip](https://github.com/Ultimaker/Cura/files/15326539/GV.pot_draft1.zip)

HellAholic, 2024-05-17:
Added a ticket for investigation, as you mentioned, it seems the 0 value is causing a problem since even a 0.0001 value improves the results.
CURA-11914

Side note, we're doing some work on the z seam, so the behavior is more in line with expectations I would say (pictures from alpha build below):
[image]
[image]

GregValiant, 2024-05-17:
Another clear cut case of "Everything affects everything".
@HellAholic  it appears that a "Max Hole Diameter" of .001 does make the wall problem go away, but a value of about 145 (and greater) causes the problem to surface again.  Good luck with that.

@Denied-X as you can see, the Cura team is aware of problems with some of these "thinner walled models".  I know we got side-tracked.  Going back to the original problem of not slicing - have you gotten it to slice?  I've sliced it about 25 times both right-side-up and upside-down and haven't had a problem.

I'll leave this open as a bug report.

Denied-X, 2025-08-06:
I lost access to the account and I can't even remember it.

Just wanted to say thanks to everyone who assisted with this, you're absolutely awesome people.