Ultimaker/Cura #19052: [5.7.0] unusual speeds

Dumpflam, 2024-05-10:
### Cura Version

5.7.0

### Operating System

Windows 10

### Printer

Ender-3 S1 PRO

### Reproduction steps

[Blind-Spot mirror attachment - Kopya.md](https://github.com/Ultimaker/Cura/files/15278460/Blind-Spot.mirror.attachment.-.Kopya.md)

!!!CHANGE .md TO .stl!!!
[profile.md](https://github.com/Ultimaker/Cura/files/15278475/profile.md)
!!!CHANGE .md TO .curaprofile!!!

hi when slicing this model with these settings the speed stays at 13mm after the short pillar is printed. dunno why.

### Actual results

low speed for no reason.

### Expected results

normal speeds.

### Add your .zip and screenshots here ⬇️

[image]

Dumpflam, 2024-05-10:
> Minimum layer time?

YUB!
do you think i should just make it 0 cuz i have a double 5015 fan setup?

GregValiant, 2024-05-12:
The slow down occurs right where the print becomes a single "island" and the layer time is just about cut in half.
If you are confident in your layer cooling setup you can drop the minimum layer time.  It may need some experimentation.  Fortunately, it's a small print.

What is going on appears to be expected behavior rather than a bug.  I'll remove the bug label but I'll leave this open for a bit so you can respond.

Dumpflam, 2024-05-12:
Thank you! Very informative. I think ill lower it a bit. Alright, cya!

12 May 2024 Pzr 05:15 tarihinde GregValiant ***@***.***> şunu
yazdı:

GregValiant, 2024-05-12:
I'll close this then.  Good luck.

closed 2024-05-12 by GregValiant