Ultimaker/Cura #19054: [5.7.0] After saving a file with supports and re-loading the file the supports are out of position

stingray-01, 2024-05-11:
### Cura Version

5.7.0

### Operating System

Windows 10 Pro

### Printer

Creality Ender-3 Pro

### Reproduction steps

Just recenlty after updating to 5.7.0 every file that I save and re open the supports are out of position. I can load an older file with no issues. I tried reverting back to 5.6.0 but it is now having the same issue if I save a new file.

1. Insert model
2. Add supports using the market place plugin  'Cylindric Custom Support 2.8.0'
3. Save the file
4. Close Cura
5. Open Cura
6. Load the file

### Actual results

Supports are out of position.

### Expected results

Supports should be in the same position as saved.

### Add your .zip and screenshots here ⬇️

[image]
[image]

GregValiant, 2024-05-14:
Thanks for the report.
When you save a file using "File | Save Project" all the models, support blockers, etc. and their locations should be saved in the file.  There would be no supports as the file because it is a "prior to slicing" save of the build plate and your settings.
When you save the Gcode generated from a slice, then the supports are part of the gcode file and cannot move around.

Looking at those images, I can't tell what is going on.  The "cyan" colored surface seems to be the surface that is resting on the build plate.  I don't see any supports there at all.  If that's a slice then it looks incomplete and kind of looks like there are model errors that are interfering with the slice.

Use the "File | Save Project" command and zip the project file and post the zip folder here.

5axes, 2024-05-15:
Custom Cylindric Support is not any more compatible with Cura due to this bug, since release 5.4 . That's why it have been removed from the Marked place.

GregValiant, 2024-05-15:
Thanks @5axes I see it now.

5axes, 2024-05-29:
Easy bug fix 😂

GregValiant, 2024-05-29:
Wait a minute @5axes  ...are we getting your plugins back?

5axes, 2024-05-29:
No unfortunatly and with this type of Github bot action .. no way to have one day a solution 😭

5axes, 2024-06-03:
Not totaly Solved with Cura 5.7.2

Ok additional support element are at the right place .. but if you save and load the project the part is not at the right place. 

[image]