Ultimaker/Cura #19056: wall order

mariusleut, 2024-05-11:
### Is your feature request related to a problem?

I'm always frustrated when the printer starts by printing all the penultimate walls and then the final walls. Since my printer isn't perfect, this ultimately results in poor outer shell quality and overall stability. It would be much better if, for instance, three wall thicknesses were defined to be printed directly in sequence.

### Describe the solution you'd like

I wish for a setting that adjusts the printing sequence so that, for example, with holes: Each hole is printed directly with all three walls, then moves to the next hole.

### Describe alternatives you've considered

While getting a better printer might be an alternative, it's not feasible for many people. I believe that many printers could benefit from this feature.

### Affected users and/or printers

I believe that many printers could benefit from achieving a better-looking print, even if the setup of the extruder, nozzle, and possibly moist filament isn't perfect. And I'm not alone in this; there are others who have searched the internet for such a solution. I recall that the Prusa Slicer performed better in this regard.

### Additional information & file uploads

In summary, my concern is about the printing sequence when there are multiple independent outer walls on a single layer, such as with holes. My wish is that each connected outer wall and inner walls are printed directly in sequence. For example, first, complete Inner Wall 1, then Inner Wall 2, followed by the Outer Shell for Hole 1, then repeat the process for Hole 2, and so on. This reduces the movement of the print head, resulting in less material loss.

GregValiant, 2024-05-12:
Have you tried enablling "Optimize Wall Printing Order"?  It was designed to allow both the inner and outer walls in an area to complete before moving to another area.
You can see here that two of the holes have been completed and that the third hole and the outer walls of the model haven't been started.
[image]

Block137, 2024-05-16:
It is bugged in some way (I'm using 5.7.0)
[image]

GregValiant, 2024-05-16:
Without a project file we have nothing to look at to investigate.