Ultimaker/Cura #19063: Unable to slice on v5.7.1 Fidget Cube with Snapmaker Artisan

sillyfrog, 2024-05-12:
### Cura Version

5.7.1

### Operating System

MacOS Sonoma (v14.4.1)

### Printer

Snapmaker Artisan

### Describe model location

Does not appear to help

### Describe your model

Yes, this works fine (with the same settings) with v5.6

### Add your .zip here ⬇️

This works OK with v5.6, then using the same model it broke with v5.7.1. (I have also delete my 5.7 config, and let it reimport from v5.6 to double check).

[Kobayashi Error.3mf.zip](https://github.com/Ultimaker/Cura/files/15285205/Kobayashi.Error.3mf.zip)

HellAholic, 2024-05-13:
It slices without issues on 5.7.1 for me. Please include your log/configuration folder.
Also note that the Snapmaker Artisan is added via plugin so it might be related to that, cannot say for sure without the configuration folder and logs.
[image]

sillyfrog, 2024-05-13:
Thanks. 

I have no idea what's happening, but things are now working OK for me now. I've reset the profile again (deleting the `5.7` folder and letting it import from `5.6`), loaded up the saved project etc, all things are now just working which weren't before. I've not updated the app (ie: still v5.7.1), so that's super confusing as it was very repeatable for me before and I couldn't figure out how to work around it.

Thanks again for your time and taking a look, and apologies for wasting your time on this one.

closed 2024-05-13 by sillyfrog