Ultimaker/Cura #19064: Brims being added, even though "none" is selected

ArcAIN6, 2024-05-12:
### Cura Version

5.5.0 - 5.7.1

### Operating System

Windows 10

### Printer

Anet ET4X

### Reproduction steps

Import model, slice.

### Actual results

notice that for some reason cura is adding brims, even though "none" is selected. 

### Expected results

one would expect there to be NO brims at all when selecting "none", or "Skirt" or literally anything OTHER than brims. 

### Add your .zip and screenshots here ⬇️

[image]
[image]
[Odd_Brims.zip](https://github.com/Ultimaker/Cura/files/15285289/Odd_Brims.zip)

GregValiant, 2024-05-12:
Thanks for the report.
That isn't a brim.  It is a support structure being generated for the 45° chamfer on the bottom of those flat models.

This is in 5.7.1 with "Support Overhang Angle" at 40° (which is less than the angle of the chamfer) and the "Support Horizontal Expansion" at 0.8.
[image]

This is your project in 5.5.0.
[image]

In both versions - changing either the "Support Overhang Angle" to 50° (greater than the chamfer angle and less than the approximate max of 63°) or the "Support Horizontal Expansion" to 0.0 will cause that first layer support structure to go away.

This happens to be in 5.7.1 with Support Horizontal Expansion at 0, and 5.5.0 reacts the same way.
[image]

So this is a settings issue rather than a bug.  I'll remove the bug label and go ahead and close this.  I hope you understand.

closed 2024-05-12 by GregValiant

ArcAIN6, 2024-05-12:
Thanks, i was really losing my mind trying to figure why it was printing those