Ultimaker/Cura #19066: Pull request check not working

Jojo09092007, 2024-05-12:
### Cura Version

Pull requests

### Operating System

Every Operating System

### Printer

All Definitions

### Reproduction steps

1. Clone Main Branch
2. Open Pull request
3. Let the bot check it
4. See it fail

### Actual results

The bot could'nt find something latest cura

### Expected results

All the checks go through

### Add your .zip and screenshots here ⬇️

Oh just check the newest pull requests.

Jojo09092007, 2024-05-12:
https://github.com/Ultimaker/Cura/pull/19053 that is my example another guy has the same problem.

Jojo09092007, 2024-05-12:
cura_resources/latest@ultimaker/latest: Not found in local cache, looking in remotes...
cura_resources/latest@ultimaker/latest: Trying with 'cura'...
ERROR: Unable to find 'cura_resources/latest@ultimaker/latest' in remotes
Error: Process completed with exit code 1. That is the error.

jellespijker, 2024-05-13:
Thanks for reporting, closing this as fixed with https://github.com/Ultimaker/Cura/commit/16a2fc336012e09eee9df0d58d75cd0866dd2cb1

closed 2024-05-13 by jellespijker