Ultimaker/Cura #19069: New variable: extruder_enabled

elgybe, 2024-05-12:
### Is your feature request related to a problem?

I need to set the extruders temperature in the start g-code script only if the extruder is enabled (I have a multiple extruder printer).

### Describe the solution you'd like

I would like to have a variable (called extruder_enabled or whatever you like) that I can call in my start g-code script:
{extruder_enabled,0} for the first extruder
{extruder_enabled,1} for the second one.
The possible results shoud be 0 or 1 (not enabled or enabled)

### Describe alternatives you've considered

The alernative I've considered is manually editing the g-code file...

### Affected users and/or printers

It will affect all multiple extruder printers.

GregValiant, 2024-05-13:
I've gone over the StartUp Gcode of numerous multi-extruder machines.  Some are "conventional" multi-extruder machines, some are multi-extruders with mixing hot ends and a shared heater, and some are IDEX machines.
From what I've seen - the best advice is to remove all mention of the extruder temperature from the StartUp Gcode and allow Cura to add the heating commands.  Cura will not heat an un-used extruder.  If you have an extruder that is disabled then by definition it is unused.
You can also set the print temperature of a disabled extruder to "0".  That way if your enabled extruder is T0 at 200 degrees and you set the disabled T1 to "0":
M104 T0 S{material_print_temperature_layer_0,0)
M104 T1 S{material_print_temperature_layer_0,1)
will come out as
M104 T0 S200
M104 T1 S0
That is something you can do today.

elgybe, 2024-05-13:
Thank you for your advice.

I prefer to add my own code for heating, since my startup sequence is complicated. I will try not to forget to set the disables extruder temperature to 0, but still waiting for the new variable ;-)