Ultimaker/Cura #19074: Slices with layers than it should.

TomTay56, 2024-05-13:
### Cura Version

5.7.1

### Operating System

Linux Mint

### Printer

Ender-3 with Skr Mini E-3 V2.0

### Reproduction steps

1. Just slice as normal.

### Actual results

5.7.1 shows 128 mm and prints 672 layers at .2 Adaptive layers is off.

### Expected results

Sliced with 5.4 and it shows 128 mm and prints 640 layers at .2 Prusa slicer also prints 640 layers at .2
[image]
[image]

### Add your .zip and screenshots here ⬇️

[image]

HellAholic, 2024-05-13:
@TomTay56 Check your shrinkage factors for the material you're using.
Generic material does not have the shrinkage factor (set at 100%) so when you slice with 0.2mm, you get 640.
If you have the ABS with shrinkage factor above 100%, you'll get more layers with all the same variables. 
That is a guess without a project file. If you would like a more concrete answer or investigation, include a project file as instructed by the report forms.

[image]
[image]

closed 2024-05-13 by TomTay56

TomTay56, 2024-05-13:
Deleted old profile. 
Made changes and saved new profile. 
All is good now.