Ultimaker/Cura #19077: G-code line number display in preview

jessegyger, 2024-05-14:
### Is your feature request related to a problem?

After making changes to a gcode file, I load the gcode file into cura and examine each step of my change. During gcode optimization, if I see a something I'd like to change while in the preview, it's difficult to know exactly where you are in the gcode to be able to make the change. 

### Describe the solution you'd like

I would like a way to tell what line of gcode I am viewing in the preview mode when I am stepping through a gcode file.  

Ideally I would really love a window that shows a few lines of the actual gcode with the ability to edit it!! But a much simpler addition of the gcode file line number would be a much appreciated addition to this software.

### Describe alternatives you've considered

The alternative is currently, count each step manually. This works for smaller models, but becomes impractical for larger models with tens of thousands of lines. 

Or buy Simplify 3d which has this feature. 

### Affected users and/or printers

It would benefit anyone who wants to locate problem areas in their gcode. 

### Additional information & file uploads

I'm visualizing the code line located at the bottom maybe above the lower left object list.. or by the play button/slider for the steps.

GregValiant, 2024-05-17:
Just thinking here..
Somewhere around this site is a request for the "Z height" to be displayed along with the layer number on the vertical slider.  Your request is similar, but I think it's harder to do.
When Cura slices it produces a gcode of the layers.  I think of it as the "raw" gcode.  When the file is saved; additional information is added to the gcode and most of that is at the beginning of the file and that creates the "finished" gcode which ends up as the gcode file.
As a consequence, I don't know if it is possible to display an accurate gcode line number in the preview because it would necessarily be of the "raw" gcode.  It could be within 20 or 30 lines (depending on things like the length of the prepend temperature lines, startup gcode, post-processor list as well as any lines added by post-processing, etc.) but I don't think it could be exact.
Opening a gcode file to preview in Cura is a different thing.  For that it might be possible to show the line number.

GregValiant, 2024-05-17:
The way I do this is so clumsy it's ridiculous.
I have several tools I've come up with that help with the triage here.  One of them is a VBA macro to read gcode files into Inventor.  Once they are in, I can select a line and get the start and end points, and I know the layer.  Then I search the gcode file for the line with the correct X's and Y's.
"Clumsy" probably isn't a strong enough word.  The system works for me but it would be much easier if the scene in Cura could display the line numbers.

gh13, 2024-05-17:
Although Cura is my slicer of choice when preparing a print there's a few tools in PrusaSlicer that I really like using. One is the gcode viewer included with the last few releases which has sliders that display line number, layer no., height & also shows the associated gcode line when scrolling. It would be great if the functionality was also incorporated into Cura but in lieu of that the PrusaSlicer gcode viewer is pretty handy