Ultimaker/Cura #19080: can't save the gcode. getting error messge: unsupported operand type(s) for +: 'float' and 'NoneType'

gioratriffon, 2024-05-15:
### Cura Version

All ver. including 5.7.1

### Operating System

Windows 10

### Printer

Ender 3

### Reproduction steps

I Made an stl file on Freecad or tinkercad and exported it to Cura. It sliced with no problem, but when I tried to save it I reseived the error message: "unsupported operand type(s) for +: 'float' and 'NoneType' " and it would not save. I tryed it with different Versions of Cura that worke with no problem previously but received the same result. Then I tryed it with files I had printe before with no problem but still I received the same error message. Then I tride a different slicer And there wase no problem. As I like Cura more then the others, I hope very much you can help me as I can't print anything with cura anymore.

### Actual results

NA See above

### Expected results

I hope ill be able to save the files

### Add your .zip and screenshots here ⬇️

[Stackable+Battery+Holders.zip](https://github.com/Ultimaker/Cura/files/15323216/Stackable%2BBattery%2BHolders.zip)

HellAholic, 2024-05-15:
Thanks for the report.
Without a project file and Cura.log file we have nothing to investigate. The instructions are on the form when you created the issue.
Packed stl models downloaded from Thingiverse does not qualify as a source for investigation unfortunately, as it does not line up with your description of the issue.

fieldOfView, 2024-05-31:
For completeness, this error was likely caused by a plugin or a post processing script. A project files would not have helped to investigate the issue, but a log file would have.

gioratriffon, 2024-06-02:
First, Thank you very much for answering me. As I didn't understand the first answer. 

Second, What is and how do I get a log file?

Giora Triffon

 

מאת: Aldo Hoeben ***@***.***> 
נשלח: יום ו 31 מאי 2024 12:20
אל: Ultimaker/Cura‏ ***@***.***>
עותק: gioratriffon‏ ***@***.***>; Author‏ ***@***.***>
נושא: Re: [Ultimaker/Cura] can't save the gcode. getting error messge: unsupported operand type(s) for +: 'float' and 'NoneType' (Issue #19080)

 

For completeness, this error was likely caused by a plugin or a post processing script. A project files would not have helped to investigate the issue, but a log file would have.

fieldOfView, 2024-06-03:
See https://github.com/Ultimaker/Cura/wiki/Reporting#cura-log

Do not try to post the cura.log file as an attachment in an email; github will remove any attachments sent with email responses. Instead press [this link](https://github.com/Ultimaker/Cura/issues/19080) and post drag te cura.log file into the textbox to add a comment.

gioratriffon, 2024-06-03:
Hi, I saved the log file here. I hope it helps.
[cura.log](https://github.com/user-attachments/files/15534054/cura.log)

gioratriffon, 2024-06-03:
Thank you, I posted the file as you asked. I hope it helps. I have to apologize for my ignorance, but maybe my age  (83) is an adequate excuse?

Giora Triffon

 

מאת: Aldo Hoeben ***@***.***> 
נשלח: יום ב 03 יוני 2024 14:31
אל: Ultimaker/Cura‏ ***@***.***>
עותק: gioratriffon‏ ***@***.***>; Author‏ ***@***.***>
נושא: Re: [Ultimaker/Cura] can't save the gcode. getting error messge: unsupported operand type(s) for +: 'float' and 'NoneType' (Issue #19080)

 

See https://github.com/Ultimaker/Cura/wiki/Reporting#cura-log

Do not try to post the cura.log file as an attachment in an email; github will remove any attachments sent with email responses. Instead press this link <https://github.com/Ultimaker/Cura/issues/19080>  and post drag te cura.log file into the textbox to add a comment.

fieldOfView, 2024-06-03:
It looks like my Z Offset plugin is to blame here. It also looks like though you have it installed, you are not using it.

You can either disable it in the marketplace (via the cogwheel icon on the top right of the marketplace window), or set the Z Offset value to a non-zero value like 0.01.

gioratriffon, 2024-06-03:
Wow you are fast. I'll uninstall it and let you know if it worked. 

 

מאת: Aldo Hoeben ***@***.***> 
נשלח: יום ב 03 יוני 2024 16:18
אל: Ultimaker/Cura‏ ***@***.***>
עותק: gioratriffon‏ ***@***.***>; Author‏ ***@***.***>
נושא: Re: [Ultimaker/Cura] can't save the gcode. getting error messge: unsupported operand type(s) for +: 'float' and 'NoneType' (Issue #19080)

 

It looks like my Z Offset plugin is to blame here. It also looks like though you have it installed, you are not using it.

You can either disable it in the marketplace (via the cogwheel icon on the top right of the marketplace window), or set the Z Offset value to a non-zero value like 0.01.

closed 2024-06-03 by fieldOfView